Emilv2/huawei_solar #2: Not Importing deprecated clients. Dependency Twisted is not Installed

jlopez77, 2020-07-24:
After a power outage the integration is failing to load.

I have re-downloaded the latest version, re-configured in configuration.yaml, restarted home assistant 20 times, but the error is still there.

**Log Details (WARNING)**
Logger: pymodbus.client.asynchronous
Source: custom_components/huawei_solar/sensor.py:4
First occurred: 4:26:23 PM (1 occurrences)
Last logged: 4:26:23 PM

Not Importing deprecated clients. Dependency Twisted is not Installed

**Log Details (ERROR)**
2020-07-24 16:26:27 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 322, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 472, in async_device_update
    await self.hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.7/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 221, in update
    self._shutdown_time = self._inverter.get("shutdown_time").value.time()
  File "/usr/local/lib/python3.7/site-packages/huawei_solar/huawei_solar.py", line 57, in get
    tmp2 = datetime.utcfromtimestamp(tmp - 60 * self._time_offset)
OverflowError: timestamp out of range for platform time_t

Is there something I can do about it? Any ideas?

jlopez77, 2020-07-24:
Ok, I've made some hacking...

Line 57 of huawei_solar.py reads original:

datetime.utcfromtimestamp(tmp - 60 * self._time_offset)

I've changed it to: 

datetime.utcfromtimestamp(1434549820776/1000)

Now, obviously, I have 

system time 2015-06-17T14:03:40.776000+00:00

wich corresponds to the datetime.utcfromtimestamp(1434549820776/1000)

but it works... 

Can you please review that 57 line? 👍

Emilv2, 2020-07-24:
You're not the first one reporting this issue, but since I was busy and didn't encounter it myself it was hard to debug.

Could your give me the offending values? What are tmp and self._time_offset? If I have examples of those I should be able to figure out what I'm doing wrong.

jlopez77, 2020-07-24:
Of course, if you teach me where the f*** are the logs 👍 

I've added:

           LOGGER.debug("%s ", self._time_offset)
            LOGGER.debug("%s ", tmp)

But i cannot see anything new in the HA logs.

Emilv2, 2020-07-24:
Hmmm, `4294967295` seems very weird since that's 80 years into the future and `1595618018` etc. seem to be correct. None of those give a `OverflowError: timestamp out of range for platform time_t` error. Maybe the last is because I'm on a 64 bit machine and you're not?

I don't understand why you get that high value, but I'll put a try except block around it.

jlopez77, 2020-07-24:
Probably it has to do with the 32bits. I'm running HA docker container on a raspberry pi 4 armv7l 32bits.

Emilv2, 2020-07-24:
I pushed a fix, does it work now?

jlopez77, 2020-07-25:
It works flawlessly now 👍 

Thanks!

closed 2020-07-25 by jlopez77