Emilv2/huawei_solar #4: Question - LUNA2000 battery

FredericMa, 2021-02-11:
Hi,

Do you know if it is also possible to read and write data from a LUNA2000 battery connected to the SUN2000L inverter?

Greetings,
Frederic

Emilv2, 2021-02-11:
There is support for batteries in the modbus interface. Which values are you interested in? https://community-openhab-org.s3.dualstack.eu-central-1.amazonaws.com/original/3X/2/e/2e6fa4cc53aedd2d8f89059cc9abffc175d59d14.pdf

There's another issue about batteries open here, I've been putting it aside for a while but will look into it.

FredericMa, 2021-02-11:
I'm sorry, I'm probably missing something but I don't find anything on batteries in the document you linked to.

Well, I'm at the moment I'm curious what data would be available.
We plan to buy a battery so I'm requesting quotes at the moment and one offer uses an inverter and battery of Huawei so I was just checking if I can get any data out of it to integrate it in Home Assistant.
It would be nice to have at least things like: status, current charge, input power, output power and temperature.

The inverter they offer is SUN2000L-5KTL.

Emilv2, 2021-02-11:
I seem to have linked the wrong document, let me find the correct one.

Emilv2, 2021-02-11:
Can't seem to find it online right now, but I luckily still had a copy on my computer: [Interface Definitions (V3.0).pdf](https://github.com/Emilv2/huawei_solar/files/5968009/Interface.Definitions.V3.0.pdf)

The fields with `[Energy storage unit]` and `[Energy storage unit 1]` are for batteries, no idea what the difference between the two is. I have no experience with using them myself.

FredericMa, 2021-02-11:
Nice! Thanks for sharing this. So this confirms that it is possible to read/write the battery parameters. That's what I was looking for in the first place. If we would buy the Huawei system I'll definitely create a PR for this repo to add this functionality.

The only difference I see between both is that the first are read only registers and the other group contains all read/write registers.

FredericMa, 2021-03-16:
Battery has been ordered and will arrive somewhere in may/june.
I'll close this issue since you provided me the info I was looking for.
Thanks again!

closed 2021-03-16 by FredericMa

ValentinoRa, 2021-03-16:
I have a SUN2000L-5KTL with a LG RESU-7H battery connected.
During last weekend I tried to read the registers.
I have some problems:
- most sensors blinks between the correct value and unavailable state, maybe some adjustments need with the delay and timeout modbus settings.
- some registers shows incorrect values (inverter efficiency, internal temperature and power factor) but it could be due to my inappropriate  register settings.
- most disappointing problem is that all battery related registers results always unavailable.
Maybe the registers data have been changed, however I suspect that it is an intentional removing.
During last months I had some discussion with Huawei technical support because of a very poor actual efficiency of the battery , mainly with low power discharge (never higher than 80%, even at high discharge power), and the auto-discharge loss is as high as 0,5-1KWh each night!
Also I noted that recently the Huawei FusionSolar app include the auto-discharge energy as returned energy (!), while previously it was not. Really a bit strange.
There is no manner to find any document wich reports the actual efficiency of the LG RESU batteries. The DC efficiency only is published, but it does not include the (energy-intensive) internal DC-DC converter consumption. If someone is able to find it, let me know, please.
I asked to Huawei andd LG technical support a document wich report the efficiency of Huawei LUNA batteries. While the support service replied quickly to any other request, no reply received about this matter, even after solicitation.