Emilv2/huawei_solar #6: Can't get your component to work

gieljnssns, 2021-02-28:
I have an Huawei SUN2000-4KTL-M1

```python
❯ python3
Python 3.9.1 (default, Dec 10 2020, 11:11:14)
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import huawei_solar
>>> inverter = huawei_solar.HuaweiSolar('192.168.89.206')
>>> inverter.get("model_name")
failed to connect to device, is the host correct?
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 169, in read_register
    response = self.client.read_holding_registers(register, length)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.89.206:502)]
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 169, in read_register
    response = self.client.read_holding_registers(register, length)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.89.206:502)]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 40, in get
    response = self.read_register(reg.register, reg.length)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 172, in read_register
    raise ConnectionException(ex)
huawei_solar.huawei_solar.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.89.206:502)]
>>>
```

```
❯ ping 192.168.89.206
PING 192.168.89.206 (192.168.89.206): 56 data bytes
64 bytes from 192.168.89.206: icmp_seq=0 ttl=255 time=2.485 ms
64 bytes from 192.168.89.206: icmp_seq=1 ttl=255 time=1.194 ms
64 bytes from 192.168.89.206: icmp_seq=2 ttl=255 time=1.252 ms
64 bytes from 192.168.89.206: icmp_seq=3 ttl=255 time=1.497 ms
64 bytes from 192.168.89.206: icmp_seq=4 ttl=255 time=1.246 ms
64 bytes from 192.168.89.206: icmp_seq=5 ttl=255 time=2.199 ms
64 bytes from 192.168.89.206: icmp_seq=6 ttl=255 time=1.943 ms
```

Emilv2, 2021-02-28:
How do you connect to the inverter? Some of them (SUN2000 series I think) will only accept modbus connections when you are connected through the inverter WiFi access point.

gieljnssns, 2021-02-28:
I'm connected through lan. (On the WiFi stick)

Emilv2, 2021-02-28:
You will need to set up a separate WiFi connection with the inverter WiFi access point to be able to connect with the modbus server.

gieljnssns, 2021-03-01:
Now I have this error
```
>>> import huawei_solar
>>> inverter = huawei_solar.HuaweiSolar("192.168.89.44")
>>> inverter.get("model_name")
could not read register value, is an other device already connected?
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 40, in get
    response = self.read_register(reg.register, reg.length)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 186, in read_register
    raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?```

gieljnssns, 2021-03-01:
```
>>> from pymodbus.client.sync import ModbusTcpClient
>>> client = ModbusTcpClient('192.168.89.44', '502')
>>> result = client.read_holding_registers(30000,15)
>>> result = client.read_holding_registers(30000,15)
>>> result = client.read_holding_registers(30000,15)
>>> result = client.read_holding_registers(30000,15)
>>> result.encode()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ModbusIOException' object has no attribute 'encode'
>>>
```

Emilv2, 2021-03-01:
Ok, seems like you are able to connect now, but reading the register fails. Do you happen to have anything else connected with the modbus server? It only supports one connection at the same time.

Do any other register addresses work? (Your device is listed as supported in the [Interface.Definitions.V3.0.pdf](https://github.com/Emilv2/huawei_solar/files/6060659/Interface.Definitions.V3.0.pdf) so `30000` should work though)

gieljnssns, 2021-03-01:
> Do you happen to have anything else connected with the modbus server? It only supports one connection at the same time.

I have the FusionSolar app

gieljnssns, 2021-03-01:
```
>>> result = client.read_holding_registers(32066,15)
>>> print(result)
Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
>>> result.encode()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ModbusIOException' object has no attribute 'encode'
>>>
```

Emilv2, 2021-03-01:
Is that the app that uses cloud connection, or the one that needs a connection with the inverter WiFi access point? The cloud one shouldn't be an issue, but the other one might interfere.

gieljnssns, 2021-03-01:
I think the cloud connection
When I turn off wifi on my phone, the app still connects
And I get this error on my Mac 
```
❯ python3
Python 3.9.1 (default, Dec 10 2020, 11:11:14)
[Clang 12.0.0 (clang-1200.0.32.27)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import huawei_solar
>>> inverter = huawei_solar.HuaweiSolar("192.168.89.44")
>>> inverter.get("model_name")
could not read register value, is an other device already connected?
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 40, in get
    response = self.read_register(reg.register, reg.length)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 186, in read_register
    raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?
>>>
```

Emilv2, 2021-03-02:
Are you sure you are connected correctly? I see the subnet has not changed. You're still on `192.168.86.0/8`. The inverter should have the address `192.168.200.1` (`200` could be something else) when you connect to it's AP.

gieljnssns, 2021-03-02:
I'm sorry, I've found it...
Now my nuc is connected to my main Lan through ethernet (like it always was) AND my nuc is connected through wifi to the inverters AP.
And now I use 
```yaml
  - platform: huawei_solar   
    host: '192.168.200.1'
    optimizers: false
    battery: false
```

Emilv2, 2021-03-02:
```
Update of sensor.sun2000_4ktl_m1 is taking over 10 seconds
```
This will be fixed once we can use async pymodbus, for now the component is just slow.

For the other issues, does it happen very often? I would just ignore it if it doesn't happen too often, I don't think I can do anything about it anyway...

gieljnssns, 2021-03-03:
It happens +/- 2 times an hour...

gieljnssns, 2021-03-04:
Since the new HA this component never starts again
```
Logger: homeassistant.loader
Source: loader.py:802
First occurred: 15:05:59 (1 occurrences)
Last logged: 15:05:59

No 'version' key in the manifest file for custom integration 'huawei_solar'. This will not be allowed in a future version of Home Assistant. Please report this to the maintainer of 'huawei_solar'
```
```
Logger: pymodbus.client.asynchronous
Source: /usr/local/lib/python3.8/site-packages/pymodbus/client/asynchronous/__init__.py:42
First occurred: 15:05:59 (1 occurrences)
Last logged: 15:05:59

Importing deprecated clients. Dependency Twisted is Installed
```
```
Logger: homeassistant.components.sensor
Source: /usr/local/lib/python3.8/asyncio/events.py:81
Integration: Sensor (documentation, issues)
First occurred: 15:06:11 (1 occurrences)
Last logged: 15:06:11

Setup of sensor platform huawei_solar is taking over 10 seconds.
```
```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:232
Integration: Sensor (documentation, issues)
First occurred: 15:07:01 (1 occurrences)
Last logged: 15:07:01

Setup of platform huawei_solar is taking longer than 60 seconds. Startup will proceed without waiting any longer.
```
```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:319
Integration: Sensor (documentation, issues)
First occurred: 15:15:24 (1 occurrences)
Last logged: 15:15:24

Timed out adding entities for domain sensor with platform huawei_solar after 500s
```

Emilv2, 2021-03-04:
I'll look into it

voctor-lab, 2021-03-05:
> I'll look into it

> Since the new HA this component never starts again
>
> ```
> Logger: homeassistant.loader
> Source: loader.py:802
> First occurred: 15:05:59 (1 occurrences)
> Last logged: 15:05:59
>
> No 'version' key in the [...]

Me too. Thanks Emilv2 for your work!!!

TBomek, 2021-03-05:
Exactly, I have the same problem. The biggest problem is that going back to the previous verse doesn't help. Any idea?

Emilv2, 2021-03-05:
Ok, give me some time. I just found out it's possible to install requirements from git, so I think I will just make it async first and see if that solves the issue.

kamilian2, 2021-03-11:
Hallo i have problem with integration. 
I have connection with inverter. 
[image]

voctor-lab, 2021-03-21:
> Ok, give me some time. I just found out it's possible to install requirements from git, so I think I will just make it async first and see if that solves the issue.

Hi! could you check it out? Thanks!!!

eagle-phoenix, 2021-03-27:
CORE Logs: 
`ERROR (MainThread) [homeassistant.components.sensor] Setup of platform huawei_solar is taking longer than 60 seconds. Startup will proceed without waiting any longer.`
nmcli Logs:
`nmcli -p -f ip4 device show wlan0
===============================================================================
                            Device details (wlan0)
===============================================================================
IP4.ADDRESS[1]:                         192.168.200.103/24
IP4.GATEWAY:                            192.168.200.1
IP4.ROUTE[1]:                           dst = 0.0.0.0/0, nh = 192.168.200.1, mt = 20600
IP4.ROUTE[2]:                           dst = 192.168.200.0/24, nh = 0.0.0.0, mt = 600
IP4.DNS[1]:                             192.168.200.1
-------------------------------------------------------------------------------
`
ping to Huawei Solar20005KTL
`~ $ ping 192.168.200.1
PING 192.168.200.1 (192.168.200.1): 56 data bytes
64 bytes from 192.168.200.1: seq=0 ttl=255 time=1.964 ms
64 bytes from 192.168.200.1: seq=1 ttl=255 time=1.978 ms
64 bytes from 192.168.200.1: seq=2 ttl=255 time=1.674 ms
`
sensors.yaml:
`- platform: huawei_solar
  host: 192.168.200.1
  optimizers: false
  battery: false`

Hassio 
core-2021.3.4
supervisor-2021.03.6
Home Assistant OS 5.12

JacoboSanchez, 2021-04-02:
Don't know if it was casual or not, but after some tests downloading (not using as they do not work with huawey) other modbus integration (SAJ MODBUS TCP)  this integration started working again.
Probably unrelated but if anyone wants to try... i have two sun2000 currently working in HA

javierin, 2021-07-14:
I currently monitor the inverter using another app via modbus-tcp. However If I try to connect HA also to the same modbus IP/port, everything collapses and it stops responding. Is this a firmware version related issue on the side of the inverter or a limitation in modbus-tcp specifications (no more than a single client expected).

Thanks!

Emilv2, 2021-07-14:
@javierin As far as I know this is a limitation in the modbus-tcp implementation on the inverter, only one client at a time can connect.

Apparently there is a firmware update for a dongle that solves some issues with modbus: https://community.openhab.org/t/reading-data-from-huawei-inverter-sun-2000-3ktl-10ktl-via-modbus-tcp-and-rtu/87670/83?u=emilv2, but I don't know anything about firmware updates for the inverter. You could  always try to contact Huawei to see if there are firmware updates for your inverter at: eu_inverter_support@huawei.com.

javierin, 2021-07-14:
Yes, I recently upgraded the dongle to a newer version, but that was a couple of months ago, I have sent support an email asking for any updates.

Thanks

Matthias54321, 2021-09-12:
Hi,

this application is exactly what I was looking for.

Unfortunately I can't get it to run.

I have a Sun2000 6 KTL M1 whose IP in the network I know.

I installed the repository via HACS as a custom repository and it is now also available as an integration under HACS.

Additionally I added the part for the sensor in configuration.yaml.

sensor:
    - platform: huawei_solar   
        host: '192.168.0.73'

But now I can't find any sensor or anything from the inverter, did I forget something?

purcell-lab, 2021-11-24:
> Hallo i have problem with integration. I have connection with inverter. [image] core-2021.3.3 supervisor-2021.03.4 docker 20.10.5

DId you get a resolution for this, I am seeing the same error..

closed 2026-11-03 by gieljnssns