Emilv2/huawei_solar #8: Can't work

kamilian2, 2021-03-11:
Hallo i have problem with integration. 
I have connection with inverter. 
[image]

purcell-lab, 2021-11-25:
> Hallo i have problem with integration. I have connection with inverter. [image]

Have you managed to resolve this issue?

I am seeing the same problem, looks like a dependant module pymodbus is not being loaded automatically.

Can we load this module manually within the HA environment?

Emilv2, 2021-11-25:
It should load automatically, I'm not sure what is going on with your install.

purcell-lab, 2021-11-25:
:It should load automatically, I'm not sure what is going on with your
install.

Thanks Emilv2,

Pretty much a HA Blue installed out if the box, so pymodbus should auto
load.

Can we add pymodbus to manifest.json or any other way to ensure it has
loaded?

Nigel, perhaps upgrade all components from Supervisor menu?

Emilv2, 2021-11-25:
You could try that, but I think the issue lies somewhere else since pymodbus is already a dependency of huawei-solar which is in manifest.json