Emilv2/huawei_solar #10: Error

Thull3, 2021-04-13:
Im suddenly getting this error..
Nothing changed what I know 
Anyone that have the same error ?? 

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:91 
Integration: Sensor (documentation, issues) 
First occurred: 6:34:04 PM (1 occurrences) 
Last logged: 6:34:04 PM

Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/huawei_solar/huawei_solar.py", line 164, in read_register
    response = self.client.read_holding_registers(register, length)
  File "/usr/local/lib/python3.8/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.8/site-packages/pymodbus/client/sync.py", line 107, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.8.1:502)]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 205, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 79, in setup_platform
    entities.append(HuaweiSolarSensor(inverter))
  File "/config/custom_components/huawei_solar/sensor.py", line 91, in __init__
    self._name = self._inverter.get("model_name").value
  File "/usr/local/lib/python3.8/site-packages/huawei_solar/huawei_solar.py", line 40, in get
    response = self.read_register(reg.register, reg.length)
  File "/usr/local/lib/python3.8/site-packages/huawei_solar/huawei_solar.py", line 167, in read_register
    raise ConnectionException(ex)
huawei_solar.huawei_solar.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.8.1:502)]

butsjoh, 2021-04-25:
I am having the same issue today as well after checking this morning. I did not change anything. I know they are doing a platform upgrade but i don't think that should interfere with the modbus integraton cause it is local. What i don't know if is that i need to be connected to to the wifi of the inverter or not because before i did not really need todo that. I just made the box connect to my own wifi and then i could approach it with the ip address my router was given it. Is this maybe not stable enough?

butsjoh, 2021-04-25:
Ok i managed to solve my issue. I had to reconnect the the inverter back to my wifi by reforcing it to reconnect and that seemed to brought the connection back in order.