Emilv2/huawei_solar #12: configuring charging etc?

wmertens, 2021-04-26:
I just got the LUNA2000 batteries and you can't configure them to discharge during the night + charge if needed before the tariff gets more expensive, to cover the morning or gray days.

Is there a way to tell the inverter to charge the batteries right now to x percent?

javierin, 2021-07-08:
Using the Huawei FusionSolar APP you can configure up to 12 (IIRC) time windows in which you can charge / discharge / export to grid. Connect to the inverter SSID and go to device provisioning using an installer account. Then follow the Quick Settings wizard and you will eventually find that scheduler.

Unfortunately I have not been able to find a way to manually trigger a charge configuration when the weather is going to be bad the following day.

wmertens, 2021-07-08:
I just noticed that in the maintenance panel of the app you can tell the battery to charge. That would work right? Presumably they're using a command over modbus?

wmertens, 2021-07-08:
Ideally, I'd be able to determine "I need x charge once the daytime period starts, based on predictions", and then during nighttime the battery would not be allowed to drop below x, and if it is below, it charges.

Emilv2, 2021-07-09:
Yes, those options seem to be available through modbus. The avaliable modbus definitions can be found in issue #18

wmertens, 2021-07-09:
Wow awesome! I see that indeed everything is available, including requesting that it charges. I'll close this, thanks!

closed 2021-07-09 by wmertens

javierin, 2021-07-13:
@wmertens i am more than interested in any piece of code that would allow writing a charging schedule to the inverter.