Emilv2/huawei_solar #15: Last version high cpu usage

gieljnssns, 2021-05-23:
When enabling this custom component I getting high cpu usage.
Enabled:

[image]

Not enabled:
[image]

Py-spy https://www.dropbox.com/s/ykcqc455v58h5yr/spy-2021.5.5.svg?dl=0

https://community.home-assistant.io/t/python3-high-cpu-usage/160012/160

drizzle1, 2021-07-02:
Any progress on this? I'm getting the same issue, but even worse in terms of CPU usage.

Emilv2, 2021-11-10:
I have released an async version of the sensor, does the issue still occur with  version 1.2.2?

gieljnssns, 2021-11-10:
I'm using the integrated modbus component now...

closed 2026-11-03 by gieljnssns