Emilv2/huawei_solar #16: inverters swapped (random) on restart HA

drizzle1, 2021-06-17:
“sensor.sun2000l_4ktl” and “sensor.sun2000l_4ktl_2” are swapped whenever HA restarts.
This messes up cards and data, graphs...
I'm guessing this can be solved by assigning the devices serial number as unique id?
https://developers.home-assistant.io/docs/config_entries_config_flow_handler/#unique-ids

JacoboSanchez, 2021-06-18:
I will provide a workaround just in case you want to try it

If you are comfortable with code editing you can try this modification i tested on a forked branch some weeks ago (not using nor maintaining it as i have a 2k and 3k so I have no name conflict). I was adding a new configuration option to prefix the entity name with this [commit](https://github.com/JacoboSanchez/huawei_solar/commit/11b5ba6c6b55f521281a9c9b3ce0eb94a3e6db59)

Also if you  want to change the entity name to the serial number and you know how to use github you can make your own fork of this project and edit sensor.py file changing [this line](https://github.com/Emilv2/huawei_solar/blob/38e794fed74a541fe2284ec6cf1a1d396bba836d/custom_components/huawei_solar/sensor.py#L124) from 
       ` self._name = self._inverter.get("model_name").value`
to 
        ` self._name = self._inverter.get("serial_number").value`

And add your fork to your HA installation (or just modify that line directly on your installation)

drizzle1, 2021-06-21:
Thanks for the workaround. I changed the line you pointed out to:
`self._name = self._inverter.get("model_name").value + self._inverter.get("serial_number").value`

so i can search for/reference to the model name as well.
That solves it for now. Thanks again.

JacoboSanchez, 2021-06-22:
I'm glad it helped... I am now making tests as I have just installed batteries in one of the inverters and found some problems getting battery information

drizzle1, 2021-06-22:
For the batteries, one thing you need to fix is:

```
if self._battery_installed:
            attributes[ATTR_STORAGE_STATUS] = self._storage_status
            attributes[ATTR_STORAGE_CHARGE_STATUS] = self._storage_charge_status
            attributes[
                ATTR_STORAGE_CHARGE_DISCHARGE_POWER
            ] = self._storage_charge_discharge_power
```
ie. you should replace the ":" by "=" signs like on last line above

luisdomg, 2021-06-22:
> I'm glad it helped... I am now making tests as I have just installed batteries in one of the inverters and found some problems getting battery information

Out of curiosity, which model did you install? I've found compatible ones quite expensive...

JacoboSanchez, 2021-06-22:
> For the batteries, one thing you need to fix is:
> 
> ```
> if self._battery_installed:
>             attributes[ATTR_STORAGE_STATUS] = self._storage_status
>             attributes[ATTR_STORAGE_CHARGE_STATUS] = self._storage_charge_status
>             attributes[
>                 ATTR_STORAGE_CHARGE_DISCHARGE_POWER
>             ] = self._storage_charge_discharge_power
> ```
> 
> ie. you should replace the ":" by "=" signs like on last line above (for all attributes)

Yeah thanks! I just made a merge request yesterday when i figured out that was the problem... now I am looking for a missing property to obtain the battery state of charge. I found the modbus memory address but it is not available from the huawei_solar python module used by this integration (and also implemented by Emil)

@luisdomg I have a 5kWh luna2000 which Huawei made for this kind of inverters so there are no compatibility issues (I did not made the installation by myself)

JacoboSanchez, 2021-06-22:
FYI I have opened the issue #18  with a workaround to obtain the state of charge for the batteries in case anyone wants that information

Emilv2, 2021-11-08:
This should be fixed in the latest version, open a new issue if you still have issues.

closed 2021-11-08 by Emilv2