Emilv2/huawei_solar #18: Add support for battery state of charge

JacoboSanchez, 2021-06-22:
After some research (as it does not seem to be documented by Huawei) I found a way to obtain the state of charge from the inverter.  [Here](https://forum.huawei.com/enterprise/en/reading-soc-soh-from-external-storage-unit-from-modbus-tcp/thread/711369-100027) I found a reply pointing to it:
`37004 - 1 register - unsigned integer - 1 decimal (so you have to divide by 10 to get the % value)  `

In record 37005 I get 1000 as value so my guess is that 37004 is the soc and 37005 is the range (so % may be (reg[37004]/reg[37005])*100... however just dividing by 10 seems safe enough and requires just one call.

As huawei-solar  python library does not return such information as named register I made a custom modification in my home assistant installation by creating a new attribute in sensor.py file this way:
`            self._storage_soc = int.from_bytes(self._inverter.read_register(37004, 1), byteorder="big")/10`

And now I am getting the percentage of charge so I can use it on my cards
[image]

I am not making a pull request as probably the correct way to add this should be to add the record to the huawei-solar python library first but I am opening this ticket to provide all the information I have

tet024, 2021-07-08:
> After some research (as it does not seem to be documented by Huawei) I found a way to obtain the state of charge from the inverter. [...]

Hello Jacobo, 
Is it possible to share you card please ?

Thanks, 
Et

JacoboSanchez, 2021-07-08:
I am using [power-wheel-card](https://github.com/gurbyz/power-wheel-card) installed through HACS. It does not seems to be in active developement and the battery information was left in beta state. You can also test [power-distribution-card](https://github.com/JonahKr/power-distribution-card) and [tesla-style-solar-power-card](https://github.com/reptilex/tesla-style-solar-power-card)

Emilv2, 2021-07-08:
Hi, thanks for the info! I will add that information, too bad I don't have a battery to poke around with. 
 
>In record 37005 I get 1000 as value so my guess is that 37004 is the **soc** and 37005 is the range (so % may be (reg[37004]/reg[37005])*100... however just dividing by 10 seems safe enough and requires just one call.

What do you mean by 'soc'?

JacoboSanchez, 2021-07-08:
I mean State of Charge,( the value of charge left on the battery). One registry seems to be the value from 0 to 1000 and the other one was giving me allways 1000 so I was just guessing that they may be related.  If you want me to give you any modbus tests output I will do it as soon as i have time to do so

tet024, 2021-07-09:
> After some research (as it does not seem to be documented by Huawei) I found a way to obtain the state of charge from the inverter. [...]

Hi, 

Can you please explain what do i need modify to read the battery charge percentage please ?

JacoboSanchez, 2021-07-09:
Sure, I have modified the file locally but now made a fork with two commits with my local modifications so you can look at the diff with the modifications.
[In this commit](https://github.com/JacoboSanchez/huawei_solar/commit/3f18ed8dd6711d9b67a8209689b902161b123c1f) adding the storage_soc attribute (also made some reordering on calls to the inverter to group related values).

In the last commit of [this fork](https://github.com/JacoboSanchez/huawei_solar) I have commented out some attributes that I don't use so there are less tcp calls to the inverter (not necessary but just to clean up the long list of attributes)

You can copy the storage_soc related lines of the first commit in your local installation, the entire file or use the fork as repo in home assistant (In this case It will be better for you to make a fork yourself as I am not developing nor maintaining it).

tet024, 2021-07-09:
> Sure, I have modified the file locally but now made a fork with two commits with my local modifications so you can look at the diff with the modifications.
> [In this [...]

I just applied the last commit  and using that now its working its working fine. Last thing regarding the cards .... are you creating sensors using the attributed from the entities or else from the care you could call the attribute something like this ?

state_attr('sensor.sun2000_3_68ktl_l1', 'storage_current_day_charge_capacity')

Emilv2, 2021-07-09:
Hi, I got the latest modbus definitions from Huawei, the extra battery definitions are in there so we don't need to guess :)

JacoboSanchez, 2021-07-09:
> I just applied the last commit and using that now its working its working fine. Last thing regarding the cards .... are you creating sensors using the attributed from the entities or else from the [...]

Not all cards work with attributes so I allways create template sensors. Also some cards work with the variation value (negative charge/positive discharge) while other cards require entity with positive value for discharge and another entity with positive value for  charge... so I ended up with a bunch of sensors with combinations for different representations of the same thing... as an example this is how I define a sensor for battery discharge value:

     storage_discharge:
        friendly_name: "Descarga batería"
        unit_of_measurement: "W"
        value_template: >-
          {% set u = state_attr('sensor.sun2000_2ktl_l1', 'storage_charge_discharge_power') | int %}
          {% if u < 0  %}
              {{ (-1*u) }}
          {% else %}
              {{ (0) | int }}
          {% endif %}`

 While for other cards I use directly the "storage_charge_discharge_power" value (also as a sensor template

JacoboSanchez, 2021-07-09:
> Hi, I got the latest modbus definitions from Huawei, the extra battery definitions are in there so we don't need to guess :)

Oh that sounds great... could you share the official description for registers 37004-37005? Are there any other interesting register to test?

Emilv2, 2021-07-09:
Yes, here's the file [Solar Inverter Modbus Interface Definitions.pdf](https://github.com/Emilv2/huawei_solar/files/6792170/Solar.Inverter.Modbus.Interface.Definitions.pdf).
37004 is the battery state of charge with a gain of 10 as you guessed, 37005 is not mentioned. There are quite a few options to monitor and regulate the charging and discharging.

tet024, 2021-07-09:
Great are you going to update the repository with your latest findings ? Keep it up you are doing a great JOB!

javierin, 2021-09-23:
There's a newer modbus definition file with tens of new registers related to LUNA2000. Please find it attached. 

[Solar Inverter Modbus Interface Definitions_202108.pdf](https://github.com/Emilv2/huawei_solar/files/7220601/Solar.Inverter.Modbus.Interface.Definitions_202108.pdf)

Emilv2, 2021-11-08:
This is supported in the latest version

closed 2021-11-08 by Emilv2