Emilv2/huawei_solar #19: Battery Attributes Not Showing

tet024, 2021-07-08:
Hello, 

Would require some help. I installed batteries today and i would like to view the parameters.  This is my config 

  - platform: huawei_solar   
    host: 'x.x.x.x'
    optimizers: true
    battery: true

In the attribute list of the invertor the storage details are missing.  If i disable the optimizers variable I see the optimizers disappear but with newly added batteries I can't see anything.  Any ideas on how to troubleshoot ?

Thanks, 
Etienne

Emilv2, 2021-07-08:
I just merged a fix for the battery attributes. Does it work now?

tet024, 2021-07-09:
> I just merged a fix for the battery attributes. Does it work now?

I Modified abit the scipt with the following and it worked. 

But let me merge and revert. 

ATTR_STORAGE_STATUS: self._storage_status,
ATTR_STORAGE_CHARGE_DISCHARGE_POWER: self._storage_charge_discharge_power,
ATTR_STORAGE_CURRENT_DAY_CHARGE_CAPACITY: self._storage_current_day_charge_capacity,
ATTR_STORAGE_CURRENT_DAY_DISCHARGE_CAPACITY: self._storage_current_day_discharge_capacity,
ATTR_STORAGE_WORKING_MODE: self._storage_working_mode,
ATTR_STORAGE_TIME_OF_USE_PRICE: self._storage_time_of_use_price,
ATTR_STORAGE_LCOE: self._storage_lcoe,
ATTR_STORAGE_MAXIMUM_CHARGING_POWER: self._storage_maximum_charging_power,
ATTR_STORAGE_MAXIMUM_DISCHARGING_POWER: self._storage_maximum_discharging_power,
ATTR_STORAGE_POWER_LIMIT_GRID_TIED_POINT: self._storage_power_limit_grid_tied_point,
ATTR_STORAGE_CHARGING_CUTOFF_CAPACITY: self._storage_charging_cutoff_capacity,
ATTR_STORAGE_DISCHARGING_CUTOFF_CAPACITY: self._storage_discharging_cutoff_capacity,
ATTR_STORAGE_FORCED_CHARGING_AND_DISCHARGING_PERIOD: self._storage_forced_charging_and_discharging_period,
ATTR_STORAGE_FORCED_CHARGING_AND_DISCHARGING_POWER: self._storage_forced_charging_and_discharging_power,

tet024, 2021-07-09:
> I just merged a fix for the battery attributes. Does it work now?

It worked :-)

Emilv2, 2021-07-15:
I'll close this issue then

closed 2021-07-15 by Emilv2