Emilv2/huawei_solar #20: after core upfrade to 7.2, having issues

akyuznet, 2021-07-15:
after core upfrade to 7.2, having issues : 

2021-07-15 14:54:02 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
await task
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 289, in update
self._grid_exporterd_energy = self._inverter.get("grid_exporterd_energy").value
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 262, in get
reg = REGISTERS[name]
KeyError: 'grid_exporterd_energy'

akyuznet, 2021-07-15:
solved: 

script owner suggested: 

Edit manifest.json in custom_components/huawei_solar/ and replace "huawei-solar>=0.1.0" with " "huawei-solar>=0.1.0,<1.0.0"`

Emilv2, 2021-07-15:
It should be fixed now in the latest release without having to edit `manifest.json`

akyuznet, 2021-07-15:
after update: 

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 291, in update
    self._grid_exported_energy = self._inverter.get("grid_exported_energy").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 39, in get
    reg = REGISTERS[name]
KeyError: 'grid_exported_energy'

Emilv2, 2021-07-15:
Ok, seems like the requirements don't get updated always. Does it work now?

mp-007, 2021-07-16:
Still the same.

Emilv2, 2021-11-08:
Does this still happen in the latest version?

closed 2021-11-08 by Emilv2