Emilv2/huawei_solar #21: after core update to 7.2, My  sensor.sun2000_15ktl_m2 entity missing.

zsgodor, 2021-07-15:
Hi Emil,
I update 2021.7.2 core My HA, but after the update, the sensor.sun2000_15ktl_m2 missing, all entity gone, please help me.
[image]

## System Health

version | core-2021.7.2
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
virtualenv | false
python_version | 3.9.5
os_name | Linux
os_version | 4.19.0-8-amd64
arch | x86_64
timezone | Europe/Budapest

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4538
Installed Version | 1.12.3
Stage | running
Available Repositories | 914
Installed Repositories | 11

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 10 (buster)
-- | --
update_channel | stable
supervisor_version | supervisor-2021.06.8
docker_version | 19.03.11
disk_total | 113.1 GB
disk_used | 38.7 GB
healthy | true
supported | true
supervisor_api | ok
version_api | ok
installed_addons | TasmoAdmin (0.15.0), Mosquitto broker (6.0.1), SQLite Web (2.2.2), Simple Scheduler (0.30.03), ESPHome (1.16.0), Portainer (1.1.0), File editor (5.2.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 2
views | 12
mode | storage

</details>

akyuznet, 2021-07-15:
script owner suggested: 

Edit manifest.json in custom_components/huawei_solar/ and replace "huawei-solar>=0.1.0" with " "huawei-solar>=0.1.0,<1.0.0"`

zsgodor, 2021-07-15:
Hi,
Now it's work, thank you for the fast support :)

closed 2021-07-15 by zsgodor