Emilv2/huawei_solar #22: After upgrade both Home Assistant and Huawei Solar issues with Battery

tet024, 2021-07-15:
Hello, 

After upgrading both HASS and the plugin I'm still having issues "Entity not found"  Though if then I set the batteries to False and Optimizer to False then the Entity would be created without any issues.  Any Ideas ?

Any ideas ?

Thanks, 
Et

tet024, 2021-07-16:
Hello, 

Good Morning, Small update from my end. The Entity becomes Not Found when I enabled the Battery to True, with the optimizers set to True the system would still work. 

The batteries where working find before the upgrade, now  when enabled Error :- Entity Not Found. Any ideas how one could solve this please ?

Thanks, 
Et

Emilv2, 2021-07-16:
Can you show what is in the logs?

tet024, 2021-07-16:
2021-07-16 10:55:26 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:319
Integration: Sensor (documentation, issues)
First occurred: 10:55:39 (1 occurrences)
Last logged: 10:55:39

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 319, in update
    self._storage_status = self._inverter.get("storage_status").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 262, in get
    reg = REGISTERS[name]
KeyError: 'storage_status'

tet024, 2021-07-16:
> Can you show what is in the logs?

Uploaded as requested. Any ideas what could be the root cause ? Do you require any further info ?

Emilv2, 2021-07-16:
That's exactly what I needed. Don't know where I was with my head but the requirements should be correct now. Can you test if it works now?

tet024, 2021-07-16:
> That's exactly what I needed. Don't know where I was with my head but the requirements should be correct now. Can you test if it works now?

Will. Check it out when I arrive home. What was the issue, if I may ask?

tet024, 2021-07-16:
> That's exactly what I needed. Don't know where I was with my head but the requirements should be correct now. Can you test if it works now?

mmmm... Didn't work 

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:291
Integration: Sensor (documentation, issues)
First occurred: 23:12:27 (1 occurrences)
Last logged: 23:12:27

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 291, in update
    self._grid_exported_energy = self._inverter.get("grid_exported_energy").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 39, in get
    reg = REGISTERS[name]
KeyError: 'grid_exported_energy'

Emilv2, 2021-07-17:
I'm working on a new version of Huawei Solar which changes a few register names for consistency, but the custom component should use the old version until that is finished. Soon I'll release a new version that hopefully also can work with asyncio so it can wait for responses in the background.

Ok, it changed so I guess at least the other version is being picked up. Looks like I still have to push that version to Pypi. You can delete line 39 with `reg = REGISTERS[name]
KeyError: 'grid_exported_energy'` for now.

I'll update it later today. Sorry for the inconvenience.

Shortblade, 2021-07-17:
Hi,
sorry to ask but where do we have to delete this line pls?
reg = REGISTERS[name] KeyError: 'grid_exported_energy'
Thx for this integration.

tet024, 2021-07-17:
> I'm working on a new version of Huawei Solar which changes a few register names for consistency, but the custom component should use the old version until that is finished. Soon I'll release a new [...]

It didn't fix it :-(  Hashed Line 39  "        # reg = REGISTERS[name]"
****

Emilv2, 2021-07-17:
I've pushed a new version and I think all KeyErrors related to battery should be fixed now

tet024, 2021-07-17:
> I've pushed a new version and I think all KeyErrors related to battery should be fixed now

Updated... Error changed though

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:338
Integration: Sensor (documentation, issues)
First occurred: 1:21:38 (1 occurrences)
Last logged: 1:21:38

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 338, in update
    self._storage_working_mode_b = self._inverter.get(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 262, in get
    reg = REGISTERS[name]
KeyError: 'storage_working_mode_b'

tet024, 2021-07-19:
Any ideas what would be the issue now ?

tet024, 2021-07-19:
I think i solved it ... Had to the delete the integration completely  restart.. turn off the invertor as it was giving error to communicate to port 502 and reinstalled it... ...  Now it seems OK

tet024, 2021-07-19:
Hello, 

I think i solved it ... Had to the delete the integration completely  restart.. turn off the invertor as it was giving error to communicate to port 502 and reinstalled it... ...  Now it seems ok... Need to check battery doing another restart.

tet024, 2021-07-19:
So, Same thing, When Enabling the batteries the following error occurs. 

Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:338
Integration: Sensor (documentation, issues)
First occurred: 12:10:59 (1 occurrences)
Last logged: 12:10:59

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 338, in update
    self._storage_working_mode_b = self._inverter.get(
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 262, in get
    reg = REGISTERS[name]
KeyError: 'storage_working_mode_b'

JacoboSanchez, 2021-07-19:
I just updated today and found also this issue. I have locally fixed it by editing sensor.py and updating 
`self._storage_working_mode_b = self._inverter.get(
                "storage_working_mode_b"
            ).value`
to
`self._storage_working_mode_b = self._inverter.get(
                "storage_unit_1_working_mode_b"
            ).value`

Emilv2, 2021-07-19:
Ok, I fixed that too

JacoboSanchez, 2021-07-19:
It's hard to make changes without being able to test them, thanks for your patience and support.

I think that you updated the attribute name for HA but not the call to your python library which is the one requiring the new name (at least from my point of view).

tet024, 2021-07-19:
Hello,

The last updated didn't solve it ... I solved it as per the recommendation of Jacobo 

This is the sensor.py which is working. 

`import voluptuous as vol
import logging
from homeassistant.components.sensor import PLATFORM_SCHEMA
from huawei_solar import HuaweiSolar, ReadException
import homeassistant.helpers.config_validation as cv
from homeassistant.const import CONF_HOST, POWER_WATT
from homeassistant.helpers.entity import Entity

_LOGGER = logging.getLogger(__name__)

DEFAULT_NAME = "Huawei Inverter"

CONF_OPTIMIZERS = "optimizers"
CONF_BATTERY = "battery"

ATTR_MODEL_ID = "model_id"
ATTR_SERIAL_NUMBER = "serial_number"

ATTR_NB_PV_STRINGS = "nb_pv_strings"
ATTR_RATED_POWER = "rated_power"
ATTR_GRID_STANDARD = "grid_standard"
ATTR_GRID_COUNTRY = "grid_country"
ATTR_DAILY_YIELD = "daily_yield"
ATTR_TOTAL_YIELD = "total_yield"
ATTR_DAY_POWER_PEAK = "day_active_power_peak"
ATTR_REACTIVE_POWER = "reactive_power"
ATTR_POWER_FACTOR = "power_factor"
ATTR_EFFICIENCY = "efficiency"
ATTR_GRID_FREQUENCY = "grid_frequency"
ATTR_GRID_VOLTAGE = "grid_voltage"
ATTR_GRID_CURRENT = "grid_current"
ATTR_LINE_VOLTAGE_A_B = "line_voltage_A_B"
ATTR_LINE_VOLTAGE_B_C = "line_voltage_B_C"
ATTR_LINE_VOLTAGE_C_A = "line_voltage_C_A"
ATTR_PHASE_A_VOLTAGE = "phase_A_voltage"
ATTR_PHASE_B_VOLTAGE = "phase_B_voltage"
ATTR_PHASE_C_VOLTAGE = "phase_C_voltage"
ATTR_PHASE_A_CURRENT = "phase_A_current"
ATTR_PHASE_B_CURRENT = "phase_B_current"
ATTR_PHASE_C_CURRENT = "phase_C_current"
ATTR_POWER_METER_ACTIVE_POWER = "power_meter_active_power"
ATTR_INPUT_POWER = "input_power"
ATTR_GRID_A_VOLTAGE = "grid_A_voltage"
ATTR_GRID_B_VOLTAGE = "grid_B_voltage"
ATTR_GRID_C_VOLTAGE = "grid_C_voltage"
ATTR_ACTIVE_GRID_A_CURRENT = "active_grid_A_current"
ATTR_ACTIVE_GRID_B_CURRENT = "active_grid_B_current"
ATTR_ACTIVE_GRID_C_CURRENT = "active_grid_C_current"
ATTR_ACTIVE_GRID_POWER_FACTOR = "active_grid_power_factor"
ATTR_ACTIVE_GRID_FREQUENCY = "active_grid_frequency"
ATTR_GRID_EXPORTED_ENERGY = "grid_exported_energy"
ATTR_GRID_ACCUMULATED_ENERGY = "grid_accumulated_energy"
ATTR_ACTIVE_GRID_A_B_VOLTAGE = "active_grid_A_B_voltage"
ATTR_ACTIVE_GRID_B_C_VOLTAGE = "active_grid_B_C_voltage"
ATTR_ACTIVE_GRID_C_A_VOLTAGE = "active_grid_C_A_voltage"
ATTR_ACTIVE_GRID_A_POWER = "active_grid_A_power"
ATTR_ACTIVE_GRID_B_POWER = "active_grid_B_power"
ATTR_ACTIVE_GRID_C_POWER = "active_grid_C_power"
ATTR_STARTUP_TIME = "startup_time"
ATTR_SHUTDOWN_TIME = "shutdown_time"
ATTR_INTERNAL_TEMPERATURE = "internal_temperature"
ATTR_DEVICE_STATUS = "device_status"
ATTR_NB_OPTIMIZERS = "nb_optimizers"
ATTR_NB_ONLINE_OPTIMIZERS = "nb_online_optimizers"
[... 289 lines cut ...]
            self._storage_charging_cutoff_capacity = self._inverter.get(
                "storage_charging_cutoff_capacity"
            ).value
            self._storage_discharging_cutoff_capacity = self._inverter.get(
                "storage_discharging_cutoff_capacity"
            ).value
            self._storage_forced_charging_and_discharging_period = self._inverter.get(
                "storage_forced_charging_and_discharging_period"
            ).value
            self._storage_forced_charging_and_discharging_power = self._inverter.get(
                "storage_forced_charging_and_discharging_power"
            ).value
            self._storage_state_of_capacity = self._inverter.get(
                ATTR_STORAGE_STATE_OF_CAPICITY
            ).value

        self._day_active_power_peak = self._inverter.get("day_active_power_peak").value
        for i in range(int(self._nb_pv_strings)):
            self._pv_strings_voltage[i] = self._inverter.get(
                f"pv_{i+1:02}_voltage"
            ).value
            self._pv_strings_current[i] = self._inverter.get(
                f"pv_{i+1:02}_current"
            ).value
`

Emilv2, 2021-07-19:
Ok yes, I was a bit too fast, I thought I already changed that one. I'll do some more cleanups later so mistakes like this should happen less.

Emilv2, 2021-07-19:
@tet024 you can put code blocks between triple ``` to make it a bit more readable
```
```

JacoboSanchez, 2021-07-19:
Working for me the new version, thanks!

Emilv2, 2021-07-19:
Great! If you can, could you try to some more of the battery attributes I added to see if they work correctly? And maybe tell which ones you would like to have in the component.

I haven't added any write function in the library yet, that will come later.

JacoboSanchez, 2021-07-19:
Currently I only have time to show you a screenshot of the storage attributes, they look ok for me

[image]

JacoboSanchez, 2021-07-19:
Maybe someone using two battery units may miss information about them... What I can do is to test accessing some unit 2 registers and see if they fail without having it installed (37738 for example with the unit 2 state of capacity)

JacoboSanchez, 2021-09-07:
> Great! If you can, could you try to some more of the battery attributes I added to see if they work correctly? And maybe tell which ones you would like to have in the component.

Now that the Home Energy system has batteries and total incremental support, it will be useful to add the charge and discharge total values. I am currently testing local changes with this... will create a pull request if it works as expected
`            self._storage_total_charge = self._inverter.get(
                "storage_unit_1_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_unit_1_total_discharge"
            ).value`