Emilv2/huawei_solar #23: after updating it doesn't work for me

docebit, 2021-07-16:
Hello
Today I had several updates to both HA and the huawei-solar sensor.
I have updated everything and now the inverter sensors are not working for me. It tells me that the entity is not available.

I have the HA updated like this:
core-2021.7.3
supervisor-2021.06.8
huawei_solar: 355ba80

In configuration, server controls, I click on verify information and it tells me that everything is correct.

tet024, 2021-07-17:
> Hello
> Today I had several updates to both HA and the huawei-solar sensor.
> I have updated everything and now the inverter sensors are not working for me. It tells me that the entity is not [...]

Hello, 

Can you share the logs of the failure? I have the same issue and had opened another thread, Emil is looking into it. If you could share the error and what option you have enabled that would be super.

docebit, 2021-07-17:
Logger: homeassistant.components.template.template_entity
Source: components/template/template_entity.py:72 
Integration: Template (documentation, issues) 
First occurred: 16 de julio de 2021 23:28:26 (2 occurrences) 
Last logged: 16 de julio de 2021 23:28:26

TemplateError('UndefinedError: 'None' has no attribute 'state'') while processing template 'Template("{{ ((states.sensor.sun2000_5ktl_l1.state | int) - (states.sensor.power_meter_active_power.state | int)) }}")' for attribute '_attr_state' in entity 'sensor.consumo_vivienda'
TemplateError('TypeError: '<' not supported between instances of 'NoneType' and 'int'') while processing template 'Template("{% if (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power') < 0 ) %} {{ (states('sensor.sun2000_5ktl_l1')|float) }} {% else %} {{ ( (states('sensor.sun2000_5ktl_l1')|float) - (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power')|float) ) }} {% endif %}")' for attribute '_attr_state' in entity 'sensor.autoconsumo_vivienda'

docebit, 2021-07-17:
Logger: homeassistant.components.sensor
Source: custom_components/huawei_solar/sensor.py:291 
Integration: Sensor (documentation, issues) 
First occurred: 16 de julio de 2021 23:28:23 (1 occurrences) 
Last logged: 16 de julio de 2021 23:28:23

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 291, in update
    self._grid_exported_energy = self._inverter.get("grid_exported_energy").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 39, in get
    reg = REGISTERS[name]
KeyError: 'grid_exported_energy'

tomeeka, 2021-07-17:
same error for me
`huawei_solar: Error on device update!

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 291, in update
    self._grid_exported_energy = self._inverter.get("grid_exported_energy").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 39, in get
    reg = REGISTERS[name]
KeyError: 'grid_exported_energy'
`

Shortblade, 2021-07-17:
Core: core-2021.7.3
Supervisor: supervisor-2021.06.8
Host OS: Home Assistant OS 6.1

Same error for me:

huawei_solar: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 432, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 601, in async_device_update
    await task
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 291, in update
    self._grid_exported_energy = self._inverter.get("grid_exported_energy").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 39, in get
    reg = REGISTERS[name]
KeyError: 'grid_exported_energy'

PimDoos, 2021-07-17:
Latest update (0.2.0) fixed this for me.

docebit, 2021-07-18:
It also works for me with the new update.  Thank you so much.

mp-007, 2021-07-18:
Now is working fine! Thanks for support

yousaf465, 2021-07-18:
> Latest update (0.2.0) fixed this for me.

how did you update it?

Emilv2, 2021-11-08:
Seems to be fixed, so I'll close this.

closed 2021-11-08 by Emilv2