Emilv2/huawei_solar #24: Add support for Home Assistant Energy integration

PimDoos, 2021-08-04:
To support the new [Home Assistant Energy](https://www.home-assistant.io/docs/energy) integration and history statistics, huawei_solar should: 
* Include a `state_class` of `measurement` for the main sensor
* Define a seperate sensor for daily_yield, which has a `last_reset` attribute to ensure it's tracked properly
https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

Techwolf12, 2021-08-05:
I did notice you don't need the daily_yield, a total_yield is also fine as long as it has a last_reset attribute on it. 
For example, my energy meter has a last_rest on epoch 0 (01-01-1970) and that works fine with the energy integration.

But support for this awesome new feature would be amazing!

Emilv2, 2021-08-06:
Doesn't look to complicated, I'll take a look at it.

inigoml, 2021-08-09:
+1. Would be great in order to have long term stats.

triplecillas, 2021-08-10:
> Doesn't look to complicated, I'll take a look at it.

Thanks in advance for your time and effort.

matheusvsv, 2021-08-16:
If anyone want to start using the new energy integration you can create a compatible sensor in your configuration.yaml
I created this sensor from the daily_yield attribute:

```
template:
  - sensor:
      - name: "Solar Daily Yield"
        state: >
          {{ state_attr('sensor.sun2000l_5ktl', 'daily_yield') | float }}
        device_class: 'energy'
        unit_of_measurement: 'kWh'
        state_class: "measurement"
        attributes:
          last_reset: "{{ now().replace(hour=0,minute=0,second=0,microsecond=0) }}"
```

If your huawei inverter doesn't have a working daily_yield attribute you can use the total_yield attribute but the last_reset on the code above need to be set to epoch time

PimDoos, 2021-08-16:
I am using something similar, however you have to make sure your inverter has it's builtin time properly synced. 

Anecdotal evidence: Mine lost its time after a power cut, which made it run 45 minutes slow. This caused the energy dashboard to report 30 kWh of solar production within one hour, at midnight, since the daily yield had not yet been reset by the inverter but the `last_reset` attribute was already set to the next day by Home Assistant.

matheusvsv, 2021-08-16:
I didn't have this issue but is something that can happen. Maybe it's best to use the total_yield, hass shouldn't have issues with this approach because the total_yield doesn't reset. Another solution is to integrate the inverter power to get kWh using the Riemann sum platform provided by hass

triplecillas, 2021-08-17:
Thaks matheusvsv, but I have one doubt.
¿What is the porpouse of setting last_reset attribute to zero?

matheusvsv, 2021-08-17:
@triplecillas because it's a requirement for the new energy integration, a statistics sensor won't show on the energy integration setup if don't have a last_reset attribute

triplecillas, 2021-08-18:
@matheusvsv now it is clear for me.
I will try it. :)

inigoml, 2021-08-24:
Tested daily yield configuration. Works perfectly.

However in order to calculate other parameters like "daily grid consumption" or "daily grid exported" something will have to be implemented. For my understanding, Energy only works properly with incremental values.... I've created several sensors for instant consumption, instant grid balance, etc....

Perhaps it will work with grid_exported_energy and grid_accumulated_energy global counters?

triplecillas, 2021-08-30:
If some of you, from Spain, have the "Plan Solar" tariffs from Iberdrola (Spanish Electricity Company), I have created an automation to change the tariff according to the plan.
[https://community.home-assistant.io/t/automation-for-plan-solar-from-iberdrola-spanish-electricity-company/334225](url)
Hope it helps.

inigoml, 2021-08-30:
Thank you, @triplecillas. I'm with SOM Energia but your automation can be adapted to my use case. With SOM I've a single tariff for my surplus energy but three different for my energy consumption .

lowey71, 2021-09-03:
Might be 2021.9.x issue, but had to change state_class to "total_increasing".  Otherwise, appears to be working well.  Thanks!

Vato9394, 2021-09-03:
template:
  - sensor:
      - name: "Solar Production"
        state: >
          {{ state_attr('sensor.sun2000l_4ktl', 'total_yield') | float }}
        device_class: "energy"
        unit_of_measurement: "kWh"
        state_class: "total_increasing"
        attributes:
          last_reset: "{{ state_attr('sensor.sun2000l_4ktl', 'system_time').replace(hour=0,minute=0,second=0,microsecond=0) }}"

Changing state_class to total increasing is not working. Can someone share his new yaml code to work with update 2021.09.x?

inigoml, 2021-09-04:
Its working with lowey71 tips. However, if you let energy to collect stats without changing state_class after updating to 2021.09.X, your counters will be broken. You will have to wait until tomorrow for stats to reset with day change. 
You can fix previous day stats by modifying sqlite database, table "statistics", but it's a bit hard to explain in this thread.

inigoml, 2021-09-04:
This is table where statistics are collected. Probably, your SUM cell is bad calculated.
You have to recalculate in order to fix history. The most time since you updated, the most work you have to do. In my case, just a couple of hours.
Take into consideration that you first have to check what are your metadata_id in statistics_medatada table. It varies depending of your sensors configuration. Just take this screenshot as a sample.

[image]

Emilv2, 2021-10-20:
@wlcrs Neat! If you start from scratch I would use the async version, which should work now. It would be great if you could get the integration in HA.

wlcrs, 2021-10-20:
@Emilv2 I encountered some errors with the async version when I was developing this integration a week ago. I'll look into fixing them, and will create a PR over there.

After, we can then look into getting it accepted in core.

Emilv2, 2021-11-07:
I have added support for the energy integration, please tell me if there are still issues.

@wlcrs There were indeed still some issues with async, in my latest sensor it works. Would you be willing to add a config flow to this sensor? I think that would be less work than rewriting your own sensor to use async.

Emilv2, 2021-11-08:
The latest version supports this now

closed 2021-11-08 by Emilv2