Emilv2/huawei_solar #27: Failed to connect to device with 5KTL and SDongleA-05

sdlafferty, 2021-08-24:
I've recently had a Huawei inverter installed, and after adding the config I get the following:

Error while setting up huawei_solar platform for sensor
August 21, 2021, 10:45:23 PM – (ERROR) Sensor (custom integration)
failed to connect to device, is the host correct?
August 21, 2021, 10:45:23 PM – (ERROR) /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py

My config looks like:
- platform: huawei_solar   
  host: '192.168.1.138'
  optimizers: false
  battery: false
  slave: 1

I have this added to my sensors.yaml file. Should it work fine from there or does it need to be in the configuration.yaml file?

sdlafferty, 2021-08-24:
This line is also in the logs:

pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(192.168.1.138:502): Connection unexpectedly closed 0.003082 seconds into read of 8 bytes without response from unit before it closed connection

claymen, 2021-08-24:
What firmware do you have on your dongle, and inverter. You need a specific version to be able to use this with the dongle, older versions can only access the modbus interface via the internal wifi connection on the inverter (e.g. the installer SSID) which is more fiddly to get access to.

sdlafferty, 2021-08-24:
Hi

I didn't think I had access to the installer portal, but I've had another look and I seem to. The dongle is at SPC119, so I'm trying to figure out how to trigger a firmware update. I can see the version from the portal, but no option to update. I have also contacted the installer of the system and they said that Huawei controls version updates, and there's no way to manually force an update. That doesn't seem true to me.

sdlafferty, 2021-08-24:
Is it possible that SPC12x is not available for the Australian market yet?

sdlafferty, 2021-08-24:
I have contacted Huawei inverter support (eu_inverter_support@huawei.com) as I found a mention on another forum to contact them to instigate firmware upgrades. Will update here as that progresses.

claymen, 2021-08-24:
> Is it possible that SPC12x is not available for the Australian market yet?

It's not officially released anywhere, SPC120 was when the modbus over the dongle came in, then was removed and then SPC122 came out, which was then removed apparently due to "issues" as well. In any case, as you are on SPC119 you can upgrade locally using the firmware files below. Follow the guide in the docs, it will walk you through how to do it via your phone and uploading the package file.

You can grab them from here.
https://easyupload.io/m/ury5gu

sdlafferty, 2021-08-25:
Huawei support gave me the same information you have; that SPC120 and SPC122 were withdrawn due to bugs, and that SPC123 should be available in the next 2 weeks. 

Did you have any issues with the manual update process, and have you experienced any of bug supposed bugs in SPC122?

claymen, 2021-08-25:
> Huawei support gave me the same information you have; that SPC120 and SPC122 were withdrawn due to bugs, and that SPC123 should be available in the next 2 weeks.
> 
> Did you have any issues with the manual update process, and have you experienced any of bug supposed bugs in SPC122?

No issues that I can see, would have been nice for Huawei to actually advise what the bugs are, whether they are even relevant. Still getting data fed into the FusionSolar cloud.
[image]

And now able to query that data locally via modbus for use in HA's energy management, though I will be honest, it's not quite as good as I had expected. I was hoping it would use more than just an hourly sort of breakdown.
[image]

But in any case, can now use this addon to query the data every 30 seconds or so which makes tracking down individual item power usage far easier. I was previously using the hacky method of connecting to the inverters internal SSID and querying modbus over TCP that way with my own modbus setup in HA, but this addon cuts down the work, and has a few other modbus addresses I didn't know even existed providing more detail.
[image]
[image]

I did have to use a template to massage the data into what HA wanted for the energy dashboard. This is what I am using, yes I know it has hard coded last_resets, but as I am using the total values that don't really reset unless you manually do it. For now this works well enough, you can for the solar yield use the daily_yield and base the last_reset off the start time of the inverter, but didn't really see a way to do this for the grid import/export, those are just growing totals, so makes more sense to just use the totals for all of them.

`template:
  - sensor:
      - name: "Solar Total Yield"
        state: >
          {{ state_attr('sensor.sun2000_5ktl_m0', 'total_yield') | float }}
        device_class: "energy"
        unit_of_measurement: "kWh"
        state_class: "measurement"
        attributes:
          last_reset: "1970-01-01T00:00:00+00:00"
  - sensor:
      - name: "Grid Exported Energy"
        state: >
          {{ state_attr('sensor.sun2000_5ktl_m0', 'grid_exported_energy') | float }}
        device_class: "energy"
        unit_of_measurement: "kWh"
        state_class: "measurement"
        attributes:
          last_reset: "1970-01-01T00:00:00+00:00"
  - sensor:
      - name: "Grid Accumulated Energy"
        state: >
          {{ state_attr('sensor.sun2000_5ktl_m0', 'grid_accumulated_energy') | float }}
        device_class: "energy"
        unit_of_measurement: "kWh"
        state_class: "measurement"
        attributes:
          last_reset: "1970-01-01T00:00:00+00:00"
`

jandeschuttere, 2021-08-25:
Happy I came across this issue, having the same problem and looks like this will help me resolve it 🙇 
Just as a heads up perhaps, the upcoming home assistant will introduce the state_class `total_increasing` which is from what I gather an answer to those `last_reset` attributes you've added to have the same result with the `measurement`. I've upgraded to the rc version of it where it is already available and it works as advertised (haven't got `gas` to work so far unfortunately but different topci)

have a look at their recently updated developers doc: https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

claymen, 2021-08-25:
> Happy I came across this issue, having the same problem and looks like this will help me resolve it 🙇
> Just as a heads up perhaps, the upcoming home assistant will introduce the state_class [...]

Yer I came across this earlier in the week, it sounds like it should deal with making it simpler in the next build.

I also found this which talked about it slightly better, and seems to confirm that the last_reset won't be needed when using total_increasing. Which will make the hacky fixed reset date I am using go away :)
https://developers.home-assistant.io/blog/2021/08/16/state_class_total/

webdados, 2021-09-23:
Sorry to hijack this issue but it seems related to my question.

So there's no way to connect the Inverter to the local network using the built-in WAN and get data from it?
A dongle is ALWAYS necessary?

claymen, 2021-09-23:
> Sorry to hijack this issue but it seems related to my question.
> 
> So there's no way to connect the Inverter to the local network using the built-in WAN and get data from it?
> A dongle is ALWAYS necessary?

Not directly, if you can connect something as a client to the inverters internal wifi network (e.g. the on the installer app uses) then you can query it via TCP Modbus over that. Previously that's how you had to do it, as the Dongle Wifi/Ethernet didn't expose TCP Modbus at all.

But it is a little tricky, because it's effectively a non-routable network, you connect to that inverter Wifi, and get either a 192.168.4.x or on newer firmware 192.168.200.x IP. The inverter is the gateway, so you can't just connect that to your LAN as it has no route back, at least not unless you used that subnet range so it'd be on the same segment. Which in a sense is how some people were doing it, adding a second wireless NIC to their HomeAssistant, having it connect to that network, and thus it can directly talk to the inverter.

The way I was doing it, was using a TL-WR702N which can act as a wifi client and bridge that to an ethernet port. On my firewall I could then drop an interface into that 192.168.200.x network, and then with some NAT effectively tell HA to talk to my firewall's internal IP on the main HA segment, but have the firewall NAT that into the inverter network. I guess the way to think of it is like a port forward on your router, except that instead of maintaining the source IP in the packet, re-writing it to be the firewalls IP in the 192.168.200.x segment, then the inverter just see's it as a local IP talking to it, and thus no routing which you can't adjust.

webdados, 2021-09-23:
But won't the inverter be connected to my wifi network as a client, after the installation, to be able to send data to their cloud and app, thus having an IP on my network?

(I still don't have it, the company that is doing the install Is still waiting for it)

claymen, 2021-09-23:
> But won't the inverter be connected to my wifi network as a client, after the installation, to be able to send data to their cloud and app, thus having an IP on my network?
> 
> (I still don't have it, the company that is doing the install Is still waiting for it)

That's the issue, the in built inverter wifi is an access point, not a client. You can only talk to it as a client device, as it's designed for the installer to connect to with the app to do the initial configuration. That's why the dongle is required if you want it to be a client on your own network, or connect to ethernet/wifi/4G to send data to the cloud.

webdados, 2021-09-23:
So in short, this device sends no information whatsoever even to a proprietary cloud/app?
The app is just for initial configuration? Damn...

claymen, 2021-09-23:
> So in short, this device sends no information whatsoever even to a proprietary cloud/app?
> The app is just for initial configuration? Damn...

As far as I understand it, that's how it works. Maybe the better way to think of it is that it's a replacement for the old school LCD panel on the front to set the unit up, or using a serial connection. But you can technically make it work if you want to tap into that network with some creative networking. I had it working this way for at least a year, till I got the updated dongle firmware to allow me to just directly talk to it.

webdados, 2021-09-23:
I'll try to get the dongle to be supplied with the inverter already.
Is the new firmware publicly available?

claymen, 2021-09-23:
> I'll try to get the dongle to be supplied with the inverter already.
> Is the new firmware publicly available?

It looks like SPC123 is now publicly listed. So long as you have access to the installer backed of the portal, or just get your installer to do it, you should be good to go.

[image]

Zolli, 2021-11-12:
If anyone get same errors ```Connection unexpectedly closed 0.003082 seconds into read of 8 bytes without response from unit before it closed connection```. This is caused because the ModbusRTU is set to "Enabled (Restricted)" in the inverter configuration, in this mode you need to spcify a single host in the inverter config that is able to initiate a connection. 

To able to reach ModbusRTU from any host you need to set the modbusRTU config to "Enabled (Unrestricted)". ANd also need SPC123 or later (SPC124 is the newest ATM)

This option is availabe in the installer config (Settings/Communication settings/Dongle Configuration)

Emilv2, 2021-11-12:
@Zolli Thanks for the information. I think I should put that in the documentation.

Zolli, 2021-11-13:
@Emilv2 Can i contribute somehow to the wiki section? Otherwise I can create a PR that adds a `docs` folder and contribute documentation and Q&A there.

Emilv2, 2021-11-15:
@Zolli I think just adding it to the readme will be best, more visible than a wiki. You can make a pull request.

malteger, 2022-01-19:
I have added most of the things mentioned here to the Readme, see the PR. Any contributions for missing things would be welcome, so that information on problems are more visible than in the issues

closed 2022-01-19 by Emilv2

zunnnn, 2022-10-09:
> > I'll try to get the dongle to be supplied with the inverter already.
> > Is the new firmware publicly available?
> 
> It looks like SPC123 is now publicly listed. So long as you have access to the installer backed of the portal, or just get your installer to do it, you should be good to go.
> 
> [image]

old post, but can you send me the files to upgrade the dongle? the link doesn't work anymore

PR #70 (2022-01-19): Add troubleshooting information for SDongle usage to Readme
I have added some troubleshooting information regarding the usage of the SDongle, as I had some problems getting it working properly. Everything is mostly collected from issues.

It might be a good idea to also include some information on using direct access, as support seems to have been removed in recent versions.

Closes #27