Emilv2/huawei_solar #28: Request for aditional atributes (energy monitor)

glarfs, 2021-08-24:
Hi,
I've a huawei monitor phisically attached to the mbus on the inverter (using data+ data- rs485 protocol). I'd like to add this device to home assistant and use the information that this device provides.

The device is [HUAWEI Smart Power Sensor DDSU666-H](https://support.huawei.com/enterprise/en/doc/EDOC1100020896/2287a95a/product-overview) [specs](https://solar.huawei.com/en-GB/download?p=%2F-%2Fmedia%2FSolar%2Fattachment%2Fpdf%2Fin%2Fdatasheet%2FSmartPowerSensor.pdf)

Should I create an issue to the huawei-solar python library?

Thanks

thydzik, 2021-10-02:
I would also like this and have a similar meter.

Emilv2, 2021-10-02:
Is this data not already in the sensor with  the `grid_exported_energy` attribute? What else are you missing?

thydzik, 2021-10-02:
Hi Emilv2, thanks for the response, my values are;
grid_exported_energy: 10056.15
grid_accumulated_energy: 4615.47
I was looking at it yesterday, and I think they are both accumulated values, but can't find what they match up with today for some reason.

I was further looking at it yesterday and I think this works for consumption and export;
production: {{ states('sensor.sun2000_5ktl_l1') }}
export: {{ state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power') }}
consumption: {{ states('sensor.sun2000_5ktl_l1')|int - state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power')|int}}

Emilv2, 2021-11-08:
Seems to be supported, open a new issue if there are some other missing attributes.

closed 2021-11-08 by Emilv2