Emilv2/huawei_solar #29: does not connect to inverter

docebit, 2021-08-31:
Hello, for a couple of days the Huawei cards have stopped working. In the system registry, in the Core section, this is what it tells me. However, I can access it from the Fusion-Solar website and it shows me the data of my installation.

----------------------------------------------------------------------------------------------
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:05 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [1900200552] Received invalid command: energy/info
2021-08-31 17:56:13 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (192.168.3.12, 502) failed: [Errno 111] Connection refused
2021-08-31 17:56:15 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (192.168.3.12, 502) failed: [Errno 111] Connection refused
2021-08-31 17:56:15 ERROR (SyncWorker_3) [huawei_solar.huawei_solar] failed to connect to device, is the host correct?
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 293, in read_register
    response = self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.3.12:502)]
2021-08-31 17:56:15 WARNING (MainThread) [homeassistant.components.broadlink.switch] The 'host' option near /config/configuration.yaml:171 is deprecated, please remove it from your configuration
2021-08-31 17:56:15 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up huawei_solar platform for sensor
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 293, in read_register
    response = self.client.read_holding_registers(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.3.12:502)]
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
    await asyncio.shield(task)
  File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/huawei_solar/sensor.py", line 115, in setup_platform
    HuaweiSolarSensor(inverter, config[CONF_OPTIMIZERS], config[CONF_BATTERY])
  File "/config/custom_components/huawei_solar/sensor.py", line 130, in __init__
    self._name = self._inverter.get("model_name").value
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
    response = self.read_register(reg.register, reg.length)
  File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 298, in read_register
    raise ConnectionException(ex)
huawei_solar.huawei_solar.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.3.12:502)]
2021-08-31 17:56:18 WARNING (MainThread) [homeassistant.setup] Setup of smartir is taking over 10 seconds.
2021-08-31 17:56:18 WARNING (MainThread) [homeassistant.components.broadlink.switch] The switch platform is deprecated, except for custom IR/RF switches. Please refer to the Broadlink documentation to catch up
2021-08-31 17:56:24 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ ((states.sensor.sun2000_5ktl_l1.state | int) - (states.sensor.power_meter_active_power.state | int)) }}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 393, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1398, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/local/lib/python3.9/site-packages/jinja2/filters.py", line 980, in do_int
    return int(value)
jinja2.exceptions.UndefinedError: 'None' has no attribute 'state'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 509, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 395, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: UndefinedError: 'None' has no attribute 'state'
2021-08-31 17:56:24 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('UndefinedError: 'None' has no attribute 'state'') while processing template 'Template("{{ ((states.sensor.sun2000_5ktl_l1.state | int) - (states.sensor.power_meter_active_power.state | int)) }}")' for attribute '_attr_state' in entity 'sensor.consumo_vivienda'
2021-08-31 17:56:24 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power') < 0 ) %}
  {{ (states('sensor.sun2000_5ktl_l1')|float) }}
{% else %} 
  {{ ( (states('sensor.sun2000_5ktl_l1')|float) - (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power')|float) ) }}
{% endif %}")
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 393, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1398, in _render_with_context
    return template.render(**kwargs)
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 1304, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.9/site-packages/jinja2/environment.py", line 925, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
TypeError: '<' not supported between instances of 'NoneType' and 'int'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 509, in async_render_to_info
    render_info._result = self.async_render(variables, strict=strict, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 395, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: TypeError: '<' not supported between instances of 'NoneType' and 'int'
2021-08-31 17:56:24 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('TypeError: '<' not supported between instances of 'NoneType' and 'int'') while processing template 'Template("{% if (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power') < 0 ) %}
  {{ (states('sensor.sun2000_5ktl_l1')|float) }}
{% else %} 
  {{ ( (states('sensor.sun2000_5ktl_l1')|float) - (state_attr('sensor.sun2000_5ktl_l1', 'power_meter_active_power')|float) ) }}
{% endif %}")' for attribute '_attr_state' in entity 'sensor.autoconsumo_vivienda'

juangarcianew, 2021-09-16:
Hi . i installed the home assistant on a raspberry and installed the huawei solar in hacs with no problem. put in configuration.yalm :
 sensor:
  - platform: huawei_solar   
    host: '192.168.68.138'
    optimizers: true
    battery: true
    slave: 0
I do not believe or detect the inverter sensor and I get these records:
2021-09-16 16:54:56 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-16 16:54:56 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-16 16:55:08 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2021-09-16 16:55:12 ERROR (SyncWorker_3) [huawei_solar.huawei_solar] could not read register value, is an other device already connected?
2021-09-16 16:55:12 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 633, in async_device_update
await task
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 333, in update
self._storage_running_status = self._inverter.get("storage_running_status").value
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
response = self.read_register(reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

my sisteme is:

## System Health

version | core-2021.9.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.6
os_name | Linux
os_version | 5.10.17-v8
arch | aarch64
timezone | Europe/Madrid

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.15.2
Stage | running
Available Repositories | 944
Installed Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.4
-- | --
update_channel | stable
supervisor_version | supervisor-2021.09.0
docker_version | 20.10.7
disk_total | 116.5 GB
disk_used | 2.4 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Terminal & SSH (9.1.3), Samba share (9.5.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 2
mode | storage

</details>
Sistema operativo del Host | Home Assistant OS 6.4
-- | --
Canal de actualización | stable
Versión del Supervisor | supervisor-2021.09.0

JacoboSanchez, 2021-09-16:
> uawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

As for my experience this error raises when there is another connection to the inverter. Modbus does only allow one client connected. For example if you have HA running you can not use FusionSolar app to directly (using the inverters wifi, not the cloud) connect the inverter. Also if you are connected (or have any other process connected) to the inverter then HA will get this exception

juangarcianew, 2021-09-16:
hi, for now it lets me connect to the inverter via wifi and I can enter the main menu. this tells me that it is not connecting to home assistant. It also happens to me with another inverter just like I have connected. in June it worked perfectly for me with two inverters on the same single-phase line. In home assistan it appeared to me as huawei_solar and another as huawei_solar2, but after the update in July it never worked again. now it doesn't even appear to me as a sensor or identity. do you speak Spanish?
What I can do? What steps can I take?
fd;  juan garcia

JacoboSanchez, 2021-09-16:
I just know that I got this error in the past connecting the inverter when I have another client connected to the inverter, but may be other causes. Unfortunately apart from this I can not guide you on how to fix it apart from trying to restart the inverters and the home assistant.  
In my case I get sometimes connection errors (not the ones you get) and I need to restart the inverter in order to be able to connect again

juangarcianew, 2021-09-19:
do you think jacobo that the reason for the no connection is that I only have the homeassistand installed on the raspberry, do I need some other program or do the installation with Docker for example?

JacoboSanchez, 2021-09-19:
No, that should not be the problem. I have also installed home assistant OS in a raspberry pi with the following configuration and it is connecting ok to both inverters:
```
sensor:
  - platform: huawei_solar
    host: "192.168.18.52"
    battery: true
    slave: 0
  - platform: huawei_solar
    host: "192.168.18.53"
    battery: false
    slave: 0
```
(they are not the same inverter model (2k and 3k) so I have no name conflict)

juangarcianew, 2021-09-20:
Hello, you have finally connected the two solar inverters but it has worked when I have put in false battery in both. If I activate the battery in 192.168.1.8, that the inverter with a Lg battery I get the following errors and does not connect that inverter.

sensor:
  - platform: huawei_solar
    host: "192.168.1.7"
    battery: false
    slave: 0
  - platform: huawei_solar
    host: "192.168.1.8"
    battery: false
    slave: 0
This is the configuration that works, 
this is the error if I put in the 192.168.1.8 battery: true
2021-09-20 16:11:18 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-20 16:11:18 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-20 16:11:31 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2021-09-20 16:11:34 ERROR (SyncWorker_2) [huawei_solar.huawei_solar] could not read register value, is an other device already connected?
2021-09-20 16:11:34 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 633, in async_device_update
await task
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 333, in update
self._storage_running_status = self._inverter.get("storage_running_status").value
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
response = self.read_register(reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

JacoboSanchez, 2021-09-20:
Mmm... that is interesting. I have no issues but I am using a Luna2000 (from Huawei). It seems that some information the integration is used is not compatible with LG.

Knowing this I can try to figure out what is happening there. Your error in sensor.py comes from line 333 which is:

`            self._storage_running_status = self._inverter.get("storage_running_status").value`

Looking at the modbus definition referenced [here](https://github.com/Emilv2/huawei_solar/issues/18#issuecomment-877246737) I can see that storage running status (register at address 37762)  is only compatible with Luna2000:

[image]

Depending on your programming skills you can try to remove such accesses (looking at the pdf there seems to be other attributes not suported with LG batteries)

juangarcianew, 2021-09-21:
hi, first I want to indicate that I have the huaweri sun2000l -5ktl, that is to say the old one and then that I have no idea of ​​python programming, I use other programming languages ​​but at an industrial level as for plc siemens. Anyway, if you tell me that I should change and where, if I can do it, since I really need from my battery to read the percentage of charge that the battery has, the charging and discharging power and also the soc.
If you can help me, I would test it on my inverter, then if you have to create a virtual sensor to calculate one that is needed, it could be done.

[Solar Inverter Modbus Interface Definitions (V3.0).pdf](https://github.com/Emilv2/huawei_solar/files/7204315/Solar.Inverter.Modbus.Interface.Definitions.V3.0.pdf)

JacoboSanchez, 2021-09-21:
I will try to help. I asked you to create a different topic so there is a specific LG battery issue that Emil can check and maybe somee day update the integration. Now we are using a non related topic from other connection problem so even if I help you this  will be hidden in the future

JacoboSanchez, 2021-09-21:
Power, SOC total values seems to not be supported by LG batteries and are also not present in the v3 definition. There are however such values for unit1... so we may try to adapt such addresses (and remove some other attributes we are not sure to be working on your environment).

I have created a branch with [some modifications](https://github.com/JacoboSanchez/huawei_solar/commit/078c84f9bfab72013faeda59c311d889092555ae). For example I have updated state of capacity
from 
```
           self._storage_state_of_capacity = self._inverter.get(
                "storage_state_of_capacity"
            ).value
```
to
```
            self._storage_state_of_capacity = self._inverter.get(
                "storage_unit_1_state_of_capacity"
            ).value
```

To check if that works you can try to update your sensor.py file from huawei_solar integration
[image]

And paste the contents of the sensor.py  file [with this version](https://raw.githubusercontent.com/JacoboSanchez/huawei_solar/lg_batteries/custom_components/huawei_solar/sensor.py). Then restart the home assistant, go to configuration.yaml and set lg_battery: true instead of battery:true for the inverter with LG batteries and restart again... then post the results 

Example:
```
- platform: huawei_solar
  host: "192.168.1.8"
  lg_battery: true
  slave: 0
```

juangarcianew, 2021-09-21:
hello jacobo and thanks for helping me. This is the result after following the steps indicated.
2021-09-21 23:05:06 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-21 23:05:06 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-21 23:05:19 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2021-09-21 23:05:24 ERROR (SyncWorker_1) [huawei_solar.huawei_solar] could not read register value, is an other device already connected?
2021-09-21 23:05:24 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 633, in async_device_update
await task
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 416, in update
self._storage_current_day_charge_capacity = self._inverter.get(
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
response = self.read_register(reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

JacoboSanchez, 2021-09-21:
Ok, that seems to be storage_current_day_charge_capacity that does not seems to be important.... I will try to remove that ones and see if the remaining ones works. You can test it first by changing 
```
            self._storage_current_day_charge_capacity = self._inverter.get(
                "storage_current_day_charge_capacity"
            ).value
            self._storage_current_day_discharge_capacity = self._inverter.get(
                "storage_current_day_discharge_capacity"
            ).value
            self._storage_maximum_charging_power = self._inverter.get(
                "storage_maximum_charging_power"
            ).value
            self._storage_maximum_discharging_power = self._inverter.get(
                "storage_maximum_discharging_power"
            ).value
```
with something not accessing such values, like:
```
            self._storage_current_day_charge_capacity = 0
            self._storage_current_day_discharge_capacity = 0
            self._storage_maximum_charging_power = 0
            self._storage_maximum_discharging_power = 0
```
And see if the remaining ones do not fail

JacoboSanchez, 2021-09-21:
I made this version removing them and just keeping soc, charge/discharge and total energy values

[sensor.py](https://raw.githubusercontent.com/JacoboSanchez/huawei_solar/main/custom_components/huawei_solar/sensor.py)

I did not test it but it is a simple change

juangarcianew, 2021-09-22:
I have put the latest version as you indicate and it comes out these errors:

huawei_solar: Error on device update!
16:11:49 – (ERROR) Sensor (integración personalizada)
could not read register value, is an other device already connected?
16:11:49 – (ERROR) /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py
Setup of sensor platform huawei_solar is taking over 10 seconds.
16:11:45 – (ADVERTENCIA) Sensor
 
2021-09-22 09:13:15 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-22 09:13:15 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

JacoboSanchez, 2021-09-22:
There is not enough information there,. In other errors you got an error trace below a line like:
` Traceback (most recent call last):`

I can not see where the error is thrown in that error lines, sorry. What I need is something like:

```
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
```

JacoboSanchez, 2021-09-22:
You can also test by yourself as there are only 4 remaining:
```
        elif self._lg_battery_installed:
            self._storage_charge_discharge_power = self._inverter.get(
                "storage_unit_1_charge_discharge_power"
            ).value
            self._storage_state_of_capacity = self._inverter.get(
                "storage_unit_1_state_of_capacity"
            ).value
            self._storage_total_charge = self._inverter.get(
                "storage_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_total_discharge"
            ).value
            
```
You can check which line does fail for you and make the change  to a 0 value instead of the self._inverter.get call to localize which ones are failing you

juangarcianew, 2021-09-22:
hello, I have rebooted before touching anything and now if the errors that you requested appear.

2021-09-22 18:04:33 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-22 18:04:33 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-09-22 18:04:47 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2021-09-22 18:04:52 ERROR (SyncWorker_5) [huawei_solar.huawei_solar] could not read register value, is an other device already connected?
2021-09-22 18:04:52 ERROR (MainThread) [homeassistant.components.sensor] huawei_solar: Error on device update!
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 431, in _async_add_entity
await entity.async_device_update(warning=False)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 633, in async_device_update
await task
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 405, in update
self._storage_total_charge = self._inverter.get(
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
response = self.read_register(reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?

JacoboSanchez, 2021-09-22:
Well that is related to 
```
            self._storage_total_charge = self._inverter.get(
                "storage_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_total_discharge"
            ).value
```
They are useful if you want to integrate battery in the home assitant energy system... but for now you can try disabling them (set to 0 instead of getting the value from inverter)... Those registers are marked as supported by LG in the last modbus document but maybe your version does not... maybe you can think about upgrading it if possible, or just live without them... but first of all try to get the basic state of charge and charge/discharge power working

juangarcianew, 2021-09-22:
good news
with this configuration:
        elif self._lg_battery_installed:
            self._storage_charge_discharge_power = self._inverter.get (
                "storage_unit_1_charge_discharge_power"
            ) .value
            self._storage_state_of_capacity = self._inverter.get (
                "storage_unit_1_state_of_capacity"
            ) .value
            self._storage_total_charge = 0
            self._storage_total_discharge = 0

you can see the inverter and the two attributes at zero en

JacoboSanchez, 2021-09-22:
That's good, as I said before in my opinion storage_total values should be compatible with your battery in newer firmware versions... but I don't know the risks on doing firmware updates can't tell you if that is a good/bad idea. 
If you want to keep it clear  you can remove those 0 assignments, however you should also remove the attribute declarations, that is, remove the lines 
```
self._storage_total_charge = 0
self._storage_total_discharge = 0
```

and also
```
            attributes[ATTR_STORAGE_TOTAL_CHARGE] = self._storage_total_charge
            attributes[ATTR_STORAGE_TOTAL_DISCHARGE] = self._storage_total_discharge
```
located here:
```
 elif self._lg_battery_installed:
            attributes[
                ATTR_STORAGE_CHARGE_DISCHARGE_POWER
            ] = self._storage_charge_discharge_power            
            attributes[
                ATTR_STORAGE_STATE_OF_CAPICITY
            ] = self._storage_state_of_capacity
            attributes[ATTR_STORAGE_TOTAL_CHARGE] = self._storage_total_charge
            attributes[ATTR_STORAGE_TOTAL_DISCHARGE] = self._storage_total_discharge
```

Note that using HACS if you update the integration all the modifications will be removed by the new version from Emil. If you want to keep your changes you may create a fork of the project, modify the sensor.py and add your fork as custom HACS repository instead  of the Emilv2's repo

JacoboSanchez, 2021-09-22:
These registers are also not in your modbus v3 pdf, but as in the working data we are using unit1 values... you may try to go back the previous version and replace the non working:
```
            self._storage_total_charge = self._inverter.get(
                "storage_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_total_discharge"
            ).value
```
with
```
            self._storage_total_charge = self._inverter.get(
                "storage_unit_1_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_unit_1_total_discharge"
            ).value
```
You may be lucky and get some data from those unit_1 values... who knows

juangarcianew, 2021-09-25:
Hello, I did the integration as you have indicated and it does not work. I would like to try the following:
            self._storage_total_charge = self._inverter.get(
                "storage_unit_1_total_charge"
            ).value
            self._storage_total_discharge = self._inverter.get(
                "storage_unit_1_total_discharge"
            ).value

now it would need to integrate these attributes to be able to form virtual sensors. in principle in the last update of modbus they are compatible.

[Energy storage unit 1] Bus voltage
[Energy storage unit 1] Bus current
[Energy storage unit 1] battery temperature

JacoboSanchez, 2021-09-25:
You can look [here](https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/huawei_solar.py#L406) the available values on the python library used by this integration. You can use those texts in the _self.inverter.get function. You can try with the variables you already have not working and use them for testing new values (if you find useful you can adapt the attribute names later). For example, you have the "storage_unit_1_total_discharge" not working. In the attached source code you can find in line 510 the following:
`    "storage_unit_1_bus_voltage": RegisterDefinitions("u16", "V", 10, 37003, 1),`
So you can try in your sensor.py file the following:
```
self._storage_total_charge = self._inverter.get(
    "storage_unit_1_bus_voltage"
).value
self._storage_total_discharge = 0

```

There is also a way to get information of a specific addres but it requires a bit more work and it seems that the values you are looking for are already in the refered library

CrouchingWorm, 2021-11-10:
Hi there, I'm having issues connecting to my inverter since 'two updates' ago, I'm currently running d22770b (hard to give you the exact version as I'm having trouble finding it).
Further more when I apply 55f1e67 it seems to corrupt my HA environment. I'm unable to restart the HA Core and other sensors become unavailable from other custom integrations.. All very odd. Only option is to restore from backup.

Love the integration!

Emilv2, 2021-11-10:
That happens when you try to fix things late at night... Things should be better now.

CrouchingWorm, 2021-11-11:
Thanks so much for the update.
I've applied the new version and all other sensors from custom integrations seem intact.
I'm still unable to connect to my inverter however.
I'll try intermittently over the next day or so to see if it comes good.
It's buggy (the inverter) it seems at times connecting via anything other than the fusion app. The inverter has 90-100% WiFi access according to my WiFi controller reporting, so connectivity shouldn't be an issue. 

Thanks again, your work is much appreciated.

CrouchingWorm, 2021-11-12:
Bingo! I've connected again. 
Thank you!

closed 2021-11-12 by Emilv2