Emilv2/huawei_solar #30: Daily Yield attribute increasing during night

Schaussi, 2021-09-07:
Hi,

my plant has been running for a few weeks now. Today I experienced an issue with the daily_yield attribute of HA's entity:

[image]

It looks like it collected about 1.4 kWh from 00:00 to 04:00am which cannot be true.

However, this value seems fo fit to my battery discharging during the night. It discharged about 2 kWh in this time range.
In addition, in the diagram you might see the yield far more increasing at 12:00, which is the time my battery went to 100% SoC.

The inverter seems to transfer the yield value to Huawei cloud correctly:
[image]

Inverter: SUN2000-10KTL-M1 (SPC133)
Dongle: SDongleA-05 (SPC122)

Please give me some information if I misunderstood anything, or if the huawei_solar integration maybe reads a wrong value from the inverter's interface.

Thank you!

JacoboSanchez, 2021-09-07:
I don't really know but... probably most of us are using L1 inverters while yours is M1. Maybe they have some differences on modbus register values/addresses?

Schaussi, 2021-09-08:
Even Sun2000 mobile app (access via inverter's AP) shows the "battery discharged" kWh included in the "daily yield" value.
I have been using it only for a few weeks now, so I cannot say if this is a bug or a feature...

Anyway, currently I am correcting the values via Node Red function like:
- sun2000 entity state changed -> 
- function to sum the attributes -> "result = yieldKwh - dischargedKwh + chargedKwh"
- send result to HA as separate entity and use this one for energy monitoring

closed 2021-09-09 by Schaussi

Emilv2, 2021-09-09:
@Schaussi It might be a bug in the firmware, maybe worth to ask the customer service (eu_inverter_support@huawei.com for the EU).

Schaussi, 2021-09-10:
@Emilv2 for my case it's okay for the moment. I am waiting for customer service responses for a few other open requests at the moment (like full FusionSolar portal installer access as plant owner, not being the installer), but will ask them in a few days.

If users of the same model are interested:
I installed SP135 today after I got it offered via current FusionSolar app (done manually via .apk file; not Sun2000 app).
The result is the same as with SP133.
For the moment I will stay with my simple calculation correction (yield + charge - discharge).

Example for the yield value today:
- FusionSolar portal: 15.11 kWh
- HA huawei_solar: 10.84 kWh
- Corrected value: 15.28 kWh