Emilv2/huawei_solar #32: Bad energy values

luzik, 2021-09-10:
Brand new SUN2000-5KTL-M1  here. All values are ok but energy is:

Grid exported energy
21474836.47
Grid accumulated energy
21474836.47

FusionSolar shows 2,46kWh

seanmccabe, 2021-09-10:
I have the same issue, posted on the HA forum and someone said was because I didn't have the meter installed.

But having said that, the app appears to know how much has been exported to the grid, so not sure that is the correct answer.

Would love to have these values working.

Happy to provide debug data or anything else needed.

Emilv2, 2021-09-10:
How can the app possibly know how much you export to the grid without a meter?

seanmccabe, 2021-09-11:
No idea, but here is a screen shot showing revenue.

[image]

luzik, 2021-09-14:
2147483647 is very special
https://en.wikipedia.org/wiki/2,147,483,647

This usually says that this param is unreliable.
Can Huawei_solar in this situation translate it to None, unknown or just delete those params?

Maybe there should be an 'extra_energy_meter'=true/false in configuration ?

inigoml, 2021-09-15:
I've the same problem. Basically, energy meter works by adding to field SUM in statistics table this formula:
SUM = ( new state - previous state ) + previous SUM

If due to outage of some kind of problem when reading inverter vaules and state becomes zero, automatically when reading again the correct value from State it will add total counters again and stats will be wrong.

[image]

Only way to fix this problem is to update manually STATE for the outage period and later recalculate SUM field from the beginning.

In order to avoid this problem, when there is some kind of problem connecting to inverter field state should be hold to previous value or NULL, at least for Total Yield, Grid exported energy and Grid accumulated energy.

Note that this problem arises if lost connection with inverter and restart HA. If you do not restart HA, state is hold at previous value.

Emilv2, 2021-09-15:
@inigoml Strange, I would expect the value being invalid when you lose connection and restart HA.

wavswavs, 2021-11-01:
Hi, sorry for the delay but I missed the notification.
There is something that I do not understand. 
I open SQLite, modify a value, the DB (home-assistant_v2.db) size goes to 0kb, and several "strange" files (H1M55I~C, HQZUGP~Y, ....) appear with elevated sizes, it is like a copy of db was created... Also, all sensors history in HA dissapear. 

What do you think might be happening?
Thanks.

inigoml, 2021-11-01:
Just install SQLite Web and forget about files.
[image]
[image]

inigoml, 2021-11-01:
This is valid if not a lot of time has passed since last statistics corruption. If many days has passed, you will have to rebuild them. Steps:

From SQLite Web export the period to fix, from first problematic day until today:
You will note that bad data is due to a "ZERO" read from inverter like this. (this is Excel).
[image]

Fix this issue and all pending data with a simple formula: 
sum = sum + (state - previous_state)
[image]

Now you can create your FIX SQL sentences by concatenating SQL commands in Excel.
[image]

[image]

Open home-assistant.db file with sqlite3 and execute sentences with ".read" command.
[image]
[image]

If you are using HomeAssistant in docker instead of server installation, you will have to extract home-assistant.db file out of docker and manipulate outside and after this, pull it again into docker. This is due to docker container does not contain sqlite3 binaries.

Ah, one final step. Just in case you are modifiying home-assistant.db offline, you will have to stop HA and remove .SHA an WAL files before starting again.
[image]

wavswavs, 2021-11-03:
I tried to fix it with SQLiteDatabaseBrowserPortable from Windows and I had the problems I mentioned. I'll try your solution step by step if it occurs again.

Now, I added Emilv2 "available" suggestion. I rely in this solution.

Thanks!.

Emilv2, 2021-11-07:
@luisdomg I have released a new version that, among other things adds the `available` attribute. I hope this will fix your issue.

wavswavs, 2021-11-11:
> I tried to fix it with SQLiteDatabaseBrowserPortable from Windows and I had the problems I mentioned. I'll try your solution step by step if it occurs again.
> 
> Now, I added Emilv2 "available" suggestion. I rely in this solution.
> 
> Thanks!.

Tonight happened again :-(
I haven't installed the new version with "available" attribute. I'll do today.

I have tried to solve the wrong data with your step by step but I make the update directly in the addon due to problems with the decimals in excel ("." - ","). It works fine but in the next period the data gets wrong again with all the accumulated data in the last period. I don´t know why, do you?
Thanks.

inigoml, 2021-11-11:
This time it's harder to maintain old stats sice meta_ids also changes, but it's possible. I finally started from scratch.
With a few adjustments due to new sensor configuration everything is working perfectly right now.

dennisgranasen, 2022-01-19:
I'm constantly getting 21474836.47 for exported and accumulated energy. I have reset those values to zero in the database and restarted but they keep returning in HA. Is there another trick I'm missing?