Emilv2/huawei_solar #33: possible to write values?

doabigcheese, 2021-09-15:
Hi,
is it somehow possible to change dynamically config data?
i am thinking of max out the yield with starting to charge the battery only if the peak of the day is reached…. this would make use of the cutoff of the grid delivery… i could imagine to reach this with setting battery charge level to 0 until it should start to load, then set to 100

cheers
bigcheese

Emilv2, 2021-09-15:
It's on my todo list, but since I don't have a battery myself I can't really test it.

JacoboSanchez, 2021-09-20:
I can make a test if you want. I don't know much about modbus, tried the following in a simple client trying  to set discharge cutoff to 13% (looking at some examples of write_register in internet):

```
    builder = BinaryPayloadBuilder(byteorder=Endian.Big)
    builder.add_16bit_uint(130)
    a = client.write_register(47082, builder.to_registers()[0]
```
but got error response

`Exception Response(134, 6, None)`

JacoboSanchez, 2021-09-23:
Tried with a modbus client gui and it seems that the error obtained is related to the lack of privileges
[image]

Emilv2, 2021-11-07:
@JacoboSanchez Maybe try to contact the Huawei support and see if they can help us with this? It's a documented feature after all.

juangarcianew, 2021-11-20:
Hello

Huawei only lets you write to registers if you use modbus rtu r485. with modbus tcp it won't let you.

javierin, 2021-12-12:
Following up on this. I sent an inquiry to Huawei and got the following response:

_Hello,
Please upgrade the SDongle to V100R001C00SPC124 and enable the Modbus TCP according to the attached guide.
You need to use Fusion Solar mobile app version 5.7.059.
The available Modbus registers for battery can be found at chapter 3.2 in the attached .
You can set Operation mode: “Time of use” based on the time when you want the battery to charge and discharge.
Please confirm you have received the files._

I have asked what specific write operations would be available over modbus tcp with this new version. I will keep you posted.

thomasdelaet, 2022-01-08:
I can confirm that write operations work if you use Modbus RTU (minor modification which you can find in this fork https://gitlab.com/tdelaet/huawei-solar). I've connected RS485 to USB adapter to pin 1 and 3 of my inverter (used for cascading inverters).

Emilv2, 2022-01-09:
@thomasdelaet These additions would be welcome!  (Was it you that asked a question about this? I can't find it anymore). If you want to you can make a pull request and we'll look over the code.

thomasdelaet, 2022-01-09:
@Emilv2 yes. it was me. I'll work on it over the next couple of weeks