Emilv2/huawei_solar #35: Help with huawei_solar.py

javierin, 2021-09-23:
Hi

I need to add time.sleep(2) to huawei_solar.py right after connection to see if I can get it to fetch data from modbus. Right now I am getting the following error in HomeAssistant:

```
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 249, in _async_setup_platform
await asyncio.shield(task)
File "/usr/local/lib/python3.9/concurrent/futures/thread.py", line 52, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/huawei_solar/sensor.py", line 122, in setup_platform
HuaweiSolarSensor(inverter, config[CONF_OPTIMIZERS], config[CONF_BATTERY])
File "/config/custom_components/huawei_solar/sensor.py", line 137, in __init__
self._name = self._inverter.get("model_name").value
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 265, in get
response = self.read_register(reg.register, reg.length)
File "/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py", line 312, in read_register
raise ReadException(message)
huawei_solar.huawei_solar.ReadException: could not read register value, is an other device already connected?
```

I am testing a modbus proxy which I can query from other devices within my LAN, but can't get this huawei_solar to work.
I have seen the error in **_/usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py_** but, damn, I can't get to find that file anywhere. How can I reach it to put the 2 seconds sleep and try to get the whole thing working?

Thanks

JacoboSanchez, 2021-09-23:
Hi, Looking at the [source code](https://gitlab.com/Emilv2/huawei-solar/-/blob/master/src/huawei_solar/huawei_solar.py#L245) it seems that it is already supported to modify the wait time, however 2 seems to be the default value.
` def __init__(self, host, port="502", timeout=5, wait=2, slave=0):`

 Try changing in sensor.py 
```
 try:
        inverter = HuaweiSolar(host=config[CONF_HOST], slave=config[CONF_SLAVE])
    except Exception as ex:
```
with 

```
 try:
        inverter = HuaweiSolar(host=config[CONF_HOST], slave=config[CONF_SLAVE], wait=2)
    except Exception as ex:
```

You can even add a configuration option in order to try different configurations

javierin, 2021-09-23:
Hi Jacobo
I managed to get it working changing slave to 0, even though I have the SDongle installed.

Thanks!