Emilv2/huawei_solar #36: I have run out of integration

docebit, 2021-10-17:
hello, for a few days this integration has not worked for me. The error message I get is:

Logger: homeassistant.helpers.template
Source: helpers/template.py:1624 
First occurred: 16 de octubre de 2021 00:35:54 (2247 occurrences) 
Last logged: 23:15:53

Template variable error: 'None' has no attribute 'replace' when rendering '{{ state_attr('sensor.sun2000_5ktl_l1', 'system_time').replace(hour=0,minute=0,second=0,microsecond=0) }}'

I don't know how to fix it. Can you give me a hand?

JacoboSanchez, 2021-10-17:
In my opinion this is a collateral error that templates are not obtaining values from the integration. Don't you have previous errors loading the huawei_solar integration? did you try to restart the inverter and checked that its IP addres has not changed from the one specified in the integration?

JacoboSanchez, 2021-10-20:
I am sorry, with that information from logs I don't know what can be the reason. In this [other topic](https://github.com/Emilv2/huawei_solar/issues/29#issue-984031636) you posted a log with connection errors after the template ones... I don't know which can cause those if connection parameters are ok.

Did you tried to connect directly with fusion app to the inverter (through the wifi, not internet) and works?
Also are your home assistant in the same network than the inverter? Port 502 will not respond if you are connecting from a different subnetwork.
Do you have an old firmware version? I think that older versions only allow modbus connection if you are directly connected to the builtin wifi network

I do not have any more ideas on what can be preventing you to connect the inverter