Emilv2/huawei_solar #40: What next after installation ---> Huawei SUN2000_Inverter

pascalppp, 2021-11-03:
Hello,

I have a problem with Huawei Solar integration installation. To be honest an installation was successful (with HACS),
I also made changes in configuration.yaml file and that's it - what should I do next :D? I can't see any changes on my dashboard. Probably solution is obvious for you, but this is my first installation of additional components for HA.

Thank You for your Ideas :)

JacoboSanchez, 2021-11-03:
This integration stores all the values as attributes, as many dashboard cards do not work with attributes you should create template sensors to extract them (I can provide an example if you want). However if you are not comfortable doing such things you can try with [this other integration](https://github.com/wlcrs/huawei_solar) that does have graphic configuration and create many sensors for use directly with the dashboard cards.

pascalppp, 2021-11-04:
Hi,

Thank You for your replay - the only entity I can see is sun.sun - please look in attchment

[image]

JacoboSanchez, 2021-11-04:
If you don't have any sun2000*** sensor then something is wrong while starting up the integration. I would restart home assistant and once it has started up check logs for any error related to the huawei_solar integration (configuration>logs)

pascalppp, 2021-11-04:
Hi,

Below You can find logs from HA - there are some problems - Huawei IP is correct and I got response with ping in CMD,
Error is visible but I don't have idea what can be a cause,
If you have any Ideas I will try to install other solar panel addon - but maybe you will notices sth in logs,
Thanks in adv for youre help!

[home-assistant.log](https://github.com/Emilv2/huawei_solar/files/7475607/home-assistant.log)

JacoboSanchez, 2021-11-04:
Error is the following:
` ModbusTcpClient(192.168.1.113:502): Connection unexpectedly closed 0.000191 seconds into read of 8 bytes without response from unit before it closed connection
`
I don't really know why the inverter isclosing the connection, however it seems similar to the error in [this issue](https://github.com/Emilv2/huawei_solar/issues/27). It seems that it had to upgrade the dongle. I know that in old firmware versions you have to be connected to the dongle's wifi to have access to 502, maybe that is your case.

I would suggest to try upgrading the firmware if you know how to do that.

pascalppp, 2021-11-04:
I've upgraded a dongle with the newest version and finally I got it: sensor.sun2000_4ktl_m1!!!:D
Thank You for your help!!!
Can you be so kind and send me a template sensor (as u said previous), which I could provide and check?
One ore time Thank You for your time and help:)

JacoboSanchez, 2021-11-04:
Sure, it is useful to understand how to create sensors with variations from data already present, you can extract the value of an attribute from other sensor and/or modify it by combining it with other values. For example you can extract the active_power value extracting the attribute power_meter_active_power from your new sensor:
```
- platform: template
   sensors:
     active_power:
       friendly_name: "Active power"
       unit_of_measurement: "W"
       value_template: >-
         {% set u = state_attr('sensor.sun2000_4ktl_m1', 'power_meter_active_power') | int %}
         {{ (u) }

```

In [this comment](https://github.com/Emilv2/huawei_solar/issues/27#issuecomment-905131529) you can see how they define sensors for energy values to integrate with the home assistant energy system.

However if you do not feel comfortable with configuration edition I would again recommend the previously mentioned alternative integration. Both use the same python library to connect to the inverter but that one has less data gathered but it has GUI configuration and extract the relevant values as sensors automatically without editing any file

pascalppp, 2021-11-06:
Unfortunetly I was happy to quickly, I have sensor.sun2000_4ktl_m1 in dev tools (after upgrading a dongle) but still errors exist:
please look at the below screen. If you don't have any idea what this time cause the problem, maybe I will try to proceed with this other installation which you mentioned before.

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/connection.py:89
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 17:53:35 (4 occurrences)
Last logged: 17:53:35

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:628
First occurred: 17:54:37 (4 occurrences)
Last logged: 17:56:07

Update of sensor.sun2000_4ktl_m1 is taking over 10 seconds

[547720334736] Received invalid command: energy/info
[image]

JacoboSanchez, 2021-11-06:
Don't really know there is no huawei_solar integration errors there. Do the sun2000_4ktl_m1 sensor or any other related derived sensor you added in the configuration returns any value? do you miss any specific one of the added templates?

pascalppp, 2021-11-08:
Hi,
I've created a template "Active power", which You send me and it is visible on the dashboard after restarting without any value, but I have added also another template: "device_status" and it shown that is on - so it is ok i believe;]
As you said - there is any errors- it is only warning that the update is pending - don't know why but it is not a case,
Anyway I will install the "other installation" to check how the configuration of the files is built. In case the questions appear I will let myself to contact you:)
Thank You

closed 2021-11-16 by pascalppp