Emilv2/huawei_solar #41: After Upgrade new verison my integration can't work

zsgodor, 2021-11-08:
Hi Emil,
After I upgrade the new version, my huawei_solar integration, "daily_yield" data is missing, can You help me please, how can i read this data my inverter?
 
My system data.:
## System Health

version | core-2021.11.1
-- | --
installation_type | Home Assistant Supervised
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 4.19.0-8-amd64
arch | x86_64
timezone | Europe/Budapest

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4566
Installed Version | 1.16.0
Stage | running
Available Repositories | 954
Installed Repositories | 16

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Debian GNU/Linux 10 (buster)
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 19.03.11
disk_total | 113.1 GB
disk_used | 24.0 GB
healthy | true
supported | failed to load: Unsupported
supervisor_api | ok
version_api | ok
installed_addons | TasmoAdmin (0.15.0), Mosquitto broker (6.0.1), SQLite Web (2.2.2), Simple Scheduler (0.31), ESPHome (2021.9.2), Portainer (1.1.0), File editor (5.2.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 4
-- | --
resources | 7
views | 14
mode | storage

</details>

Evka2k, 2021-11-08:
new sensors (sensor.daily_yield and sensor.total_yield) introduced with all required attributes for energy integration. You probably should use those instead of old sensor *sun2000* attributes.

zsgodor, 2021-11-08:
I use this.:
inverter_napitermeles:
         friendly_name: "Napi Termelés"
         unit_of_measurement: "kWh"
         value_template: '{{ states.sensor.sun2000_15ktl_m2_hv20c0067666.attributes.daily_yield }}'
What wrong?

Emilv2, 2021-11-08:
The new sensor does not have the attribute daily yield anymore, it's it's own sensor now. No template needed.

Evka2k, 2021-11-08:
No more attribute "daily_yield" for sun2000 sensor available. Changed to additional sensor "sensor.daily_yield"

Try this:
value_template: '{{ states.sensor.daily_yield.state_with_unit }}'

inigoml, 2021-11-08:
Same problem here.
Something is wrong after update. It's the integration itself...
[image]

I've removed all my sensors but integration basic config.
[image]

Evka2k, 2021-11-08:
> Same problem here. Something is wrong after update. It's the integration itself... [image]
> 
> I've removed all my sensors but integration basic config. [image]

Update to new version. 51f7e93
https://github.com/Emilv2/huawei_solar/commit/51f7e93ecb68c4d8c667220738735a0aad6f9b47
this patch fixes your problem.

inigoml, 2021-11-08:
Yep. I'm afraid upgraded between these two commits. :-)
[image]

Updating right now. ;-)

zsgodor, 2021-11-08:
"sensor.daily_yield"
"sensor.total_yield"
It works. :)
Thank You the fast support. :)

closed 2021-11-08 by zsgodor

PimDoos, 2021-11-08:
@Emilv2  Wouldn't it make more sense to call the sensor `sensor.sun2000_{model}_daily_yield`? This avoids ambiguity, as I have multiple sources of energy yield in my Home Assistant.

Emilv2, 2021-11-08:
Yes that was the plan, but there were some issues with implementing it so I dropped it temporarily. I've thought of a better way to implement it.

inigoml, 2021-11-08:
By the way, a collateral effect of these changes is that since sensors changes, they are created with new metadata id at statistics table. So, you will loose previous stats unless you update statistics table with new IDs from statistics_meta table. In particular, daily_yield or total_yield.

Emilv2, 2021-11-08:
I have updated the daily_yield (and others) naming so they are unique. I've also added the `unique_id` attribute so if you want to you can edit the name in the UI back to the original name for statistics (At least, I think that statistics don't use the unique id but the entity name).