Emilv2/huawei_solar #42: Missing field after update to 1.2.0

leeyuentuen, 2021-11-09:
after updating the component, it seems the naming has been changed for the device.
but also some fields is missing like:
- daily_yield
- total_yield

leeyuentuen, 2021-11-09:
it seems they are now separated component

mistrals, 2021-11-09:
Battery information still unavailable
[image]

leeyuentuen, 2021-11-09:
I'll reopen this issue

Emilv2, 2021-11-09:
I assume they worked before? Can you show me the logs?

mistrals, 2021-11-09:
`2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: model_id
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: model_name
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: serial_number
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: rated_power
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: nb_pv_strings
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: grid_code
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] created inverter
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] added entities `

This is what i can see in HA logs.

Is possible that i have changed battery controller and now inverter see two controller? Can you try to modify code and try to cicle battery section?

Emilv2, 2021-11-09:
What do you mean by cicle? I have just added a new version that has more debug info (and throttles some of the reconnection attempts). What info does that give?

Emilv2, 2021-11-09:
@mistrals I'll close this and we'll discuss you issue in #43.

closed 2021-11-09 by Emilv2