Emilv2/huawei_solar #43: after component update new battery created sensor are in unknown state

mistrals, 2021-11-09:
[image]

mistrals, 2021-11-10:
Con you log all the raw value readed from inverter?

Emilv2, 2021-11-10:
What logs do you get with the latest version (just pushed it)?

mistrals, 2021-11-10:
this is what i see in HA logs
`2021-11-09 17:23:09 DEBUG (MainThread) [custom_components.huawei_solar.sensor] Setup Huawei Inverter
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: model_id
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: model_name
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: serial_number
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: rated_power
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: nb_pv_strings
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get sensor static attribute: grid_code
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] created inverter
2021-11-09 17:23:12 DEBUG (MainThread) [custom_components.huawei_solar.sensor] added entities`

this is what i see in Huawei Web Portal.
The first Battery controller is the old controller no more connected
The second one is in use

[image]

Emilv2, 2021-11-10:
Better with 1.2.2?

mistrals, 2021-11-10:
yes, now sensor is with value.
can i ask why "Storage state of capacity" is disappeared?

Emilv2, 2021-11-10:
It's still there as an attribute, no? I could make it an entity

mistrals, 2021-11-11:
no, is no more present as attribute

mistrals, 2021-11-11:
ok with 1.3 now work, thanks

closed 2021-11-11 by mistrals