Emilv2/huawei_solar #44: After update frequent timeout with impact on sensor

maury77, 2021-11-09:
After update frequently timeout are present

....
 Update of sensor.sun2000l_3ktl is taking over 10 seconds
.....
custom_components.huawei_solar.sensor] could not get register 'grid_code': asyncio timeout

[custom_components.huawei_solar.sensor] could not get register 'internal_temperature': asyncio timeout

[custom_components.huawei_solar.sensor] could not get register 'active_power': asyncio timeout

.....

Emilv2, 2021-11-09:
I have noticed that too, I've added some throttling in version 1.2.1, let's see what that gives.

Emilv2, 2021-11-11:
The errors are gone for me in the latest version, does it work for you as well or do you still have issues with timeout errors?

adischmidt, 2021-11-12:
Running v1.2.3 and I am also getting errors and can't see the device.

Logger: pymodbus.client.asynchronous.async_io
Source: /srv/homeassistant/lib/python3.8/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:306
First occurred: 10:25:51 AM (10 occurrences)
Last logged: 10:26:45 AM

Failed to connect: [Errno 111] Connect call failed ('x.x.x.x', 502)

pjgrun, 2021-11-12:
Updated to version 439c23c  and I am now getting from time to time following error: 
2021-11-12 10:26:06 ERROR (MainThread) [custom_components.huawei_solar.sensor] could not get register 'active_power': asyncio timeout

Emilv2, 2021-11-12:
If it's always active power I think I know what's going on. The modbus implementation on the inverter needs some time between connection and the first request. I do this in my package, but sometimes (if the delay between requests is too long?) It is required again. Unfortunately I don't think we can see if this is necessary, so either I always put a delay be before the first (active power) request or we have this issue. I don't think adding the delay is really an issue since the component is async now.

Emilv2, 2021-11-12:
I can't reproduce this, when I increase `SCAN_INTERVAL` I see there is a disconnect and a reconnect, but no asyncio timeout. Maybe it only happens when the disconnect happens right at the time when the request is made?

```
2021-11-12 23:46:39 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.                                                                                    
2021-11-12 23:46:39 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.                                                                                    
2021-11-12 23:46:39 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.0.123:502.                                                                             
2021-11-12 23:48:40 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol lost connection.                                                                                    
2021-11-12 23:48:40 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Protocol made connection.                                                                                    
2021-11-12 23:48:40 INFO (MainThread) [pymodbus.client.asynchronous.async_io] Connected to 192.168.0.123:502.   
```

The pymodbus code is also not very clear to me so I would have to do a bit of trial and error to find a good solution even if I could reproduce the issue.

pjgrun, 2021-11-12:
Yes, it was only active power. It appears to be OK now. The component is async now? In the custom_component/huawei_solar folder I can still see the two files: sensor.py and sensor_async.py.

Emilv2, 2021-11-12:
`sensor_async.py` is obsolete, I forgot to remove it. The sensor is completely async now.

pjgrun, 2021-11-15:
Great news it is completely async now. Thank you very much for clarifying and congratulations for great achievement.  
On the other side, I can confirm that I have not had any issue with huawei_solar in the last days. It appears that when I had the errors, my HA system must not have been very stable. I started to see  deficient response time in other components as well. Fortunately, I could recover normal behaviour just by installing a previous backup. All and in particular your huawei_solar integration is now working perfectly.