Emilv2/huawei_solar #45: Fails to load

claymen, 2021-11-10:
After updating to 55f1e67 I am getting no entity or sensor setup, and with the sensor added I get the following error when restarting the core. Running under HASS OS, core-2021.11.2

[image]

Example of the previously working config entry.
`  - platform: huawei_solar
    host: x.x.x.x
    optimizers: false
    battery: false
    slave: 1`

Emilv2, 2021-11-10:
That should be fixed now.

claymen, 2021-11-10:
> That should be fixed now.

Awesome, all working. Much appreciated for the fast update. Let me know if you need any other details pulled.

closed 2021-11-10 by claymen