Emilv2/huawei_solar #46: No storage info of Luna2000 after recent updates

Y4nnickB, 2021-11-10:
Dear @Emilv2 ,

Since the last 2 updates of the integration I'm not seeing any information of my batteries.
Is this a known problem? I'm not seeing any errors in the logs...
These attributes are empty:
Storage running status -
Storage current day charge capacity -
Storage current day discharge capacity -
Storage working mode a -
Storage unit 1 working mode b -
Storage time of use price -
Storage lcoe -
Storage maximum charging power -
Storage maximum discharging power -
Storage power limit grid tied point -
Storage charging cutoff capacity -
Storage discharging cutoff capacity -
Storage forced charging and discharging period -
Storage forced charging and discharging power -
Storage state of capacity -

I'm using these attributes in templates to trigger some automations and stuff...
Thanks!

Emilv2, 2021-11-10:
Better with 1.2.2?

Y4nnickB, 2021-11-10:
@Emilv2 I just updated to 1.2.2 and the battery related entities are now working!
The attributes from my starting post are still blank...
But with the charge-discharge entity working I'm able to edit my templates and automations I think..
Thanks!

Emilv2, 2021-11-10:
Ok, attributes should be fixed in 1.2.3

Y4nnickB, 2021-11-11:
Hey @Emilv2, everything is fixed in 1.2.3 indeed...
Thanks a lot!

closed 2021-11-11 by Emilv2