Emilv2/huawei_solar #47: dont work

agmuz, 2021-11-10:
error dont work 439c23c

Emilv2, 2021-11-10:
Please provide a bit more context. What doesn't work? Can you show any logs?

lolopepes, 2021-11-11:
For me too not working from last updates

agmuz, 2021-11-11:
version 439c23c no integration visible, no attributes, I restored a copy of 2485325 work ok, I don't have a log :(

Emilv2, 2021-11-11:
Configuration -> logs shows nothing related to huawei_solar?

agmuz, 2021-11-11:
2021-11-11 20:25:37 WARNING (SyncWorker_5) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

Emilv2, 2021-11-11:
Nothing else?

What if you set the logger to debug for the huawei_solar component?

```
logger:
  default: error
    logs:
      custom_components.huawei_solar: debug
```

agmuz, 2021-11-11:
I cannot use this entry, the configuration file does not accept it

Emilv2, 2021-11-11:
Sorry, I made a small mistake in the formatting, this should work:
```
logger:
  default: info
  logs:
    custom_components.huawei_solar: debug
```

agmuz, 2021-11-11:
where is this file?

agmuz, 2021-11-11:
2021-11-11 20:56:19 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration huawei_solar which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-11 20:56:19 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration mikrotik_router which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain logger took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up http
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up system_log
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain system_log took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 0.1 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 1: {'websocket_api', 'config', 'frontend', 'webhook', 'hassio', 'lovelace', 'api', 'person', 'auth', 'onboarding', 'image', 'analytics', 'cloud', 'system_log', 'http', 'search', 'device_automation'}
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up websocket_api
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain websocket_api took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up config
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up webhook
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain webhook took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up hassio
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up api
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain api took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up auth
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain auth took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setting up device_automation
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain device_automation took 0.0 seconds
2021-11-11 20:56:20 INFO (MainThread) [homeassistant.setup] Setup of domain config took 0.5 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up image
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up analytics
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up search
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain search took 0.0 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.3 seconds
2021-11-11 20:56:21 INFO (SyncWorker_7) [homeassistant.loader] Loaded alexa from homeassistant.components.alexa
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain analytics took 0.0 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain image took 0.0 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up cloud
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up person
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain cloud took 0.2 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain person took 0.1 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up onboarding
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.helpers.storage] Migrating onboarding storage from 3 to 4
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain onboarding took 0.0 seconds
2021-11-11 20:56:21 INFO (SyncWorker_1) [homeassistant.loader] Loaded panel_custom from homeassistant.components.panel_custom
2021-11-11 20:56:21 INFO (SyncWorker_6) [homeassistant.loader] Loaded device_tracker from homeassistant.components.device_tracker
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setting up frontend
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.setup] Setup of domain frontend took 0.0 seconds
2021-11-11 20:56:21 INFO (MainThread) [homeassistant.components.http] Now listening on port 8123
2021-11-11 20:56:24 INFO (MainThread) [homeassistant.setup] Setup of domain hassio took 3.5 seconds
2021-11-11 20:56:24 INFO (SyncWorker_5) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor
2021-11-11 20:56:24 INFO (MainThread) [homeassistant.setup] Setting up binary_sensor
2021-11-11 20:56:24 INFO (MainThread) [homeassistant.setup] Setup of domain binary_sensor took 0.0 seconds
2021-11-11 20:56:24 WARNING (MainThread) [homeassistant.components.template.sensor] The 'entity_id' option near /config/configuration.yaml:250 is deprecated, please remove it from your configuration
2021-11-11 20:56:24 WARNING (MainThread) [homeassistant.components.template.sensor] The 'entity_id' option near /config/configuration.yaml:257 is deprecated, please remove it from your configuration
[... cut ...]
2021-11-11 20:56:33 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get register: pv_02_voltage
2021-11-11 20:56:33 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get register: pv_02_current
2021-11-11 20:56:33 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get register: daily_yield_energy
2021-11-11 20:56:34 DEBUG (MainThread) [custom_components.huawei_solar.sensor] get register: accumulated_yield_energy
2021-11-11 20:56:34 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.huawei_solar entity: sensor.sun2000_3ktl_l1_hv2150017624

Emilv2, 2021-11-11:
I see no problem, you should have a huawei_solar sensor. The sensors have changed names, they now include the serial number so that you can have multiple inverters without issues. Maybe that is your problem?

agmuz, 2021-11-11:
platform: huawei_solar
    host: 'ip'
after the change it will be:
  - platforms: HV2150000001
    host: 'ip'?

Emilv2, 2021-11-11:
No, the sensor has changed names, not the config. If your sensor was called `sensor.sun2000l_3_68ktl` before, it's
`sensor.sun2000l_3_68ktl_12345678901234567890` now.

agmuz, 2021-11-11:
ok

agmuz, 2021-11-11:
please update the documentation

agmuz, 2021-11-11:
works perfect

Emilv2, 2021-11-11:
Done. Great, I'll close this issue then.

closed 2021-11-11 by Emilv2

lolopepes, 2021-11-11:
> No, the sensor has changed names, not the config. If your sensor was called `sensor.sun2000l_3_68ktl` before, it's
> `sensor.sun2000l_3_68ktl_12345678901234567890` now.

Thanks!

agmuz, 2021-11-11:
TX 👍