Emilv2/huawei_solar #49: No data after latest update

erikwachters, 2021-11-13:
Hi,

The sensor is not there and this is all over the log:
WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 113] Connect call failed ('192.168.1.249', 502)

Emilv2, 2021-11-13:
If you don't use HACS you can just copy the correct commit. If you use HACS I don't know, I have never used HACS.

There didn't change anything fundamental between the versions, so I suspect your issue lies somewhere else. If not that's an issue that should be fixed instead of just not updating.

Emilv2, 2021-11-13:
Ok, strange, I can't find this warning `Connect call failed` at all in the pymodbus code.

erikwachters, 2021-11-14:
Ok I think it's solved by removing __pycache__ from the folder and restart HASS.

closed 2021-11-14 by erikwachters