Emilv2/huawei_solar #53: Meaning state and attributes

johanvh83, 2021-12-01:
This morning an electrician commissioned my brand new Huawei SUN2000-4.6KTL-L1 with 10 kWh battery (2 x 5 kWh).
I installed successfully your custom integration and I have the State and 69 other attributes.
Can you tell me where I can find a description or the meaning of these 70 values?
I checked the user manual from Huawei but couldn't find any information.
Thanks in advance.

JacoboSanchez, 2021-12-01:
Data obtained from this integration is the one provided by the inverter reading modbus tcp registers. You can find their definition in the [Modbus Interface Definitions](https://github.com/Emilv2/huawei_solar/files/6792170/Solar.Inverter.Modbus.Interface.Definitions.pdf) document. There is no real description but for some registers you have some info about the values. For example for power_meter_active_power:
[image]

You don't really have the entity and 69 attributes as this integration provides some of the old attributes as entities that can be directly used by HA, you can read it in the integration description:
[image]

For most common uses of the integration (power flows and HA energy system) the most useful are:

**Energy**: 

- entities total_yield, storage_total_charge, storage_total_discharge and 
- attributes: grid_accumulated_energy, grid_exported_energy

**Power**:
- entities: integration state (active_power), storage_charge_discharge_power
- attributes: input_power, power_meter_active_power 

You can look to [this comment in other issue](https://github.com/Emilv2/huawei_solar/issues/52#issuecomment-981061373) for some more info on those values.

johanvh83, 2021-12-01:
Thank you very much. This was exactly I was looking for!
Great job!