Emilv2/huawei_solar #54: Is it possible to select a downgraded version via HACS?

mwantia, 2021-12-02:
As the title implies, but is it possible to install a older version of this repository via HACS?

I already tried to reinstall it after version 1.0.0 all of my statistics under the tab broken.

The are either completely unreliable with only 2 or 3 entires a day showing any values or other values not showing up at all.

Everything worked before I upgraded it by mistake and only now was able to get to this problem.

JacoboSanchez, 2021-12-03:
Hacs does not have an option to specify versión afaik. However you can make a fork of this project, reset It to the desired versión and remove EmilV2 integration and add to HACS the URL of yout fork.

ibl129, 2021-12-05:
Same wish over here. Everything worked great. The inverter briefed frequently its updates but now it stops with frequently updating after a 15 minutes after restarting home assistant. I would like to downgrade as well…

mwantia, 2021-12-06:
Thanks @JacoboSanchez.

I will make a fork out of this project and try to get it to work after I have found some time for it.
Will update the results to this issue after I have figured everything out.