Emilv2/huawei_solar #56: Solar production doesn't

johanvh83, 2021-12-04:
Installed the integration. Got all attributes but I'm wondering where to find the solar production.

Based on older threads I found here, I added in my configuration.yaml:
template:
  - sensor:
      - name: "Solar Production"
        state: >
          {{ state_attr('sensor.sun2000_4_6ktl_l1_hv2180152184', 'total_yield') | float }}
        device_class: "energy"
        unit_of_measurement: "kWh"
        state_class: "measurement"
        attributes:
          last_reset: "{{ state_attr('sensor.sun2000_4_6ktl_l1_hv2180152184', 'system_time').replace(hour=0,minute=0,second=0,microsecond=0) }}"

The sensor exists but the reading is 0:
[image]

Emilv2, 2021-12-04:
The actual production in kW is the state of the sensor. There are sensors for the battery power (kW) and accumulated energy (kWh).

I think HA energy dashboard uses kWh because using power does only work correctly if you can sample all (power in, power out, production power and storage in/out power) at the exactly the same time. If you have different devices for this, this isn't really feasible. This isn't really an issue if you use the energy sensors.

johanvh83, 2021-12-05:
Ok. thanks for your answer

JacoboSanchez, 2021-12-06:
Home assistant energy panel is meant to work with energy (KWh). If you want to have power flows in W (for kW you should create templates dividing by 1000) you should manually configure some specific card in a dashboard like:
[tesla-style-solar-power-card](https://github.com/reptilex/tesla-style-solar-power-card)
[power distribution card](https://github.com/JonahKr/power-distribution-card)
[power wheel card](https://github.com/gurbyz/power-wheel-card)
or some other out there in HACS

johanvh83, 2021-12-06:
@JacoboSanchez thanks for your answer and the links to the specific cards.
It starts to provide the correct information to these cards and so far this is not completely clear.
If I am correct:
input_power = actual solar panel production in Watts
from my DSMR integration I can get information about the net (grid) delivered and injected power to the network in Watts
Where can I find the actual power charge/discharge to the battery?
If I'm not mistaken this infomation is not available in the attrributes, only percentage of battery (sensor.pv_batt_capacity) and total daily charged (storage_current_day_charge_capacity) and discharged (storage_current_day_discharge_capacity) are available attributes.
If I'm right is it possible to add the actual battery power as attribute to this project?