Emilv2/huawei_solar #57: Unreliable readings?

johanvh83, 2021-12-05:
Compared the solar production today on the FusionSolar website and the State of the sensor.
It seems there is a big difference:
Fusionsolar
[image]
Grafana in Home Assistant
[image]
I'm able to give many expamples but the pictures are clear; there are many differences.
Should I 'believe' Fusionsolar or is Grafana more reliable?

Emilv2, 2021-12-05:
I'm not sure, they could be a bit different depending on the sample rate and smoothing they use, but this looks excessive. Do the daily yields differ?

johanvh83, 2021-12-05:
The HA daily yield entity shows 7,12 kWh
[image]

While FusionSolar shows 7.03 kWh
[image]

That's not that much difference.

johanvh83, 2021-12-06:
Hi, I'm still investigating and observed, in my opinion, something to improve.
The state of the sensor is not only the solar power production but also the battery charge/discharge is part of this result.

Today it's a cloudy day. Fusionsolar gives a solar power production of about 160 watts.
My battery delivers 500 watts to my house, as a result the state of the sensor is 660 watts.

Any idea if you can rectify this?

[image]
[image]

JacoboSanchez, 2021-12-06:
I think you are using the wrong value then. From my understanding input power attribute is the DC current being produced as solar power. Entity value is the active power which is the AC current going out from the inverter to house/grid.
If the installation does not have batteries both values match (more or less). If you have batteries, they are in the DC part of the installation so current charging them go directly from solar production and not going through the inverter. Also if batteries are discharging, the power sums up to the input power so output active power of the inverter  can be higher than the solar production.

TLDR: Use input power as solar production value

johanvh83, 2021-12-06:
Thanks for your answer. You are right, the input_power attribute is the solar production in Watts.
The State is combination of solar production and battery charge/discharge.
Also some other attributes are not clear for me. I asked for clarification, you send me a document but it's not clarifying.
Do you know which atttribute shows the actual battery charge/discharge power in Watts?

JacoboSanchez, 2021-12-06:
Well I also explained that and It and posted energy/ power/attributes/entities. It is also under the integration README. storage_charge_discharge_power is not an attribute but a separate entity. It takes negative/positive values depending on charge/discharge state

Emilv2, 2021-12-06:
> Hi, I'm still investigating and observed, in my opinion, something to improve. The state of the sensor is not only the solar power production but also the battery charge/discharge is part of this result.
> 
> Today it's a cloudy day. Fusionsolar gives a solar power production of about 211 watts. My battery delivers 629 watts to my house, as a result the state of the sensor is 840 watts.
> 
> Any idea if you can rectify this?

It's not so simple, see wlcrs/huawei_solar/issues/1. In my opinion the only real solution is Huawei fixing this in the firmware. If enough people complain they might fix it...