Emilv2/huawei_solar #58: SPC124

Marchellos, 2021-12-11:
Hello all,

Has anyone got Huawei Solar sensor with SPC124 working?

skenhall, 2021-12-12:
Yes, but only with one of my two cascaded inverters (but that is a separate issue). 
With SPC 124 there is a setting to enable Modbus TCP, as it is now disabled by default: https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/789585-100027

Marchellos, 2021-12-12:
Thanks for your comment.

I upgraded to SPC124, but still I can't do step 8. 
This option is not available..

I have now approached the helpdesk of Huawei if they want to solve this problem for me.
I don't think it's because of the inverter firmware.

skenhall, 2021-12-12:
> Thanks for your comment.
> 
> I upgraded to SPC124, but still I can't do step 8. This option is not available..
> 
> I have now approached the helpdesk of Huawei if they want to solve this problem for me. I don't think it's because of the inverter firmware.

Ok, but please also check that you have the latest app version (as it is a new configuration parameter not previously available in the app). I'm using the latest version for iOS that is 5.7.059 and then the parameter is visible to be set in step 7 and 8.

Marchellos, 2021-12-12:
I have installed the latest version of the FusionSolar app.
But it has now become dark here (NL) and I have to wait until tomorrow.
If there is no sunlight you can apparently not do anything even with the presence of mains voltage.

javierin, 2021-12-12:
I have this enabled and restricted to my monitoring IP. I will try to figure out how to send write requests via modbus tcp to the inverter. I am really interested in some scripted interaction with the sun2000

Marchellos, 2021-12-13:
I just got a response from the Huawei helpdesk:

_I found that in this manual there is information that you need the latest firmware for inverter also (V100R001C00SPC139).
Maybe this is the reason why you don’t have this option. Do you have also the latest version of app?_

So upgrade first and check again tomorrow.

javierin, 2021-12-13:
I am trying with a simple 
`result = c.write_register(47088, 8, unit=0)`

This should change the maximum charge level of the battery and is displayed as a rw value in the documentation. 

However I get the following response:
`Exception Response(134, 6, None)`

Jacobo, in a different issue, explained this was caused by a permissions issue, although I don't know if he was able to overcome it and how he got to the _0x80 No permission_ from that response.

javierin, 2021-12-14:
So, regarding SPC124 and latest version of FusionSolar Android App. Yes, It can definitely be enabled,  even without sunlight. I enabled it last weekend and just changed settings when the inverter is still in standby mode (no hibernation when idle).

[image] [image]

Marchellos, 2021-12-14:
Thank you for your comment,

If there is no sunlight, the inverter switches off completely for me.
Then I can't connect locally, not even upgrade remotely.
I didn't see an option to change this either.
At this point I still don't have the option to enable modbus-tcp (only available on android?).
Maybe it depends on the converter type?

Inverter:
SUN2000-17KTL-M0
V100R001C00SPC139

dongle:

SDongleA-05
V100R001C00SPC124

Install account Fusion Solar : Yes.

And I did the latest upgrade on my Iphone:

The Smart PV Management System will be upgraded. The iPhone users need to download the latest version (5.7.059 or later) from the App Store. For Android users, perform the upgrade as prompted. If the FusionSolar app is not upgraded, some functions may be unavailable.

javierin, 2021-12-14:
This is my environment:
SUN2000-5KTL-L1 V200R001C00SPC115
SDongleA-05 V100R001C00SPC124
Android app latest version

Marchellos, 2021-12-14:
I am now waiting for a response from the Huawei helpdesk.
They usually respond quite quickly.

JacoboSanchez, 2021-12-15:
Do you know if the modbus-tcp configuration requires an installer account?

javierin, 2021-12-15:
> Do you know if the modbus-tcp configuration requires an installer account?

I definitely log into the inverter using an installer Account. You can find the specific settings in Settings (Ajustes) > Communication configuration > Dongle parameter settings > Modbus-TCP

Marchellos, 2021-12-15:
> Weet u of de modbus-tcp-configuratie een installatieaccount vereist?

No idea, I'm using an installer account.
But this option (step 8) is not available to me..

https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/789585-100027?page=2

Maybe it's the iPhone app and I have to use an android.

rufik, 2021-12-17:
> No idea, I'm using an installer account.
> But this option (step 8) is not available to me..

I had this issue as well, but on Android. Latest FusionSolar app.
But I had reset app (clear cache & data) and tried again - and this option just appeared :) So try to reinstall app and start on fresh again.

Marchellos, 2021-12-18:
Just tried again with a friend's phone. And this time with an android. Unfortunately no difference option is still not available.
I log in with the installer account.
The Huawei helpdesk has moved the problem to a higher level.

javierin, 2021-12-29:
I am rolling back inverter and dongle versions. I upgraded to inverter fw117 and sdongle 126 and i was getting no responses from the SUN2000.

inigoml, 2022-01-06:
Hi. Regarding this issue, Huawei has blocked modbus/TCP in latest upgrade at firmware level via integrated wifi. In my case, SUN2000-4KTL-L1 has no modbus access after upgrading from SPC114 to 117. Huawei support confirmed that this is by default and that I must use a SDongle FE in order to recover this access or downgrade to 114 (that I've already done).

inigoml, 2022-01-06:
[image]

motoleon, 2022-02-21:
> Hi. Regarding this issue, Huawei has blocked modbus/TCP in latest upgrade at firmware level via integrated wifi. In my case, SUN2000-4KTL-L1 has no modbus access after upgrading from SPC114 to 117. [...]

I upgradert mine to 117 and now modbus does not function. How can i downgrade to 114??? please!!!!

javierin, 2022-02-21:
I have successfully jumped back and forth firmware versions using fusionsolar > Plants > Device > Upgrade Management.
You can ask inverter support email for the required firmware version and they will probably send it to you as well so you can update it using the mobile app.

motoleon, 2022-02-22:
I need SUN2000LV200R001C00SPC114_package.zip and i cant download it because i am not seller. I have not permision to download, i am a user that purchase the inverter. I login at huawei but not permisions to download firmwares...

motoleon, 2022-02-22:
Ok. I just do it. SPC114 running and modbus on my LAN and Home Assistant reading data. Ok. A lot of troubles but...finally all ok.

inigoml, 2022-02-22:
You need to register your inverter in order to download software. No need
to be installer but device has to be registered.
Beside published software version, there are other included in FusionSolar
app and also support has newer versions. At this moment I've SPC114, SPC117
and SPC119 available.