Emilv2/huawei_solar #60: Solar yield is reduced by the battery charge

killee, 2021-12-14:
In my understanding the solar yield is the energy that the inverter gets from the PV.
The yield that I get is with the substraktion of the energy that flows into the battery.
My yield of today:
[image]
Battery energy:
[image]
And the expression on fusion:
[image]

The problem is if I use it for the energy visualization of HA it did not match...
[image]

I try to solve this by a template:
     - name: "PV2 PV Production Yield"
        unique_id: pv2_pv_production_yield
        unit_of_measurement: "kWh"
        icon: mdi:solar-power
        state: "{{ states('sensor.total_yield_hvXXXX') | float(0) + states('sensor.storage_total_charge_hvXXXX') | float(0)}}"
        device_class: energy
        state_class: total_increasing

But to solve this robust a zero handling has also to add...
In my opinion this is also not the perfect way.

Any change to handle this in the integration?

johanvh83, 2021-12-22:
I used your template for the PV Solar production but it doesn't works:
[image]
Compared to FusionSolar:
[image]

killee, 2021-12-22:
It seems to be not that easy as expected...

I think I have solved this issue. 
The attribute gives the DC Power. I integrate it with HA, than template it.

template:
  - sensor:
      - name: "PV2 Power DC"
        unique_id: pv2_pv_input_power
        unit_of_measurement: "W"
        icon: mdi:solar-power
        state: "{{ state_attr('sensor.sun2000_10ktl_m1_sun2000_12_hvXXX', 'input_power') | float(0) }}"

      - name: "PV2 DC input Yield" #use this in the energy dashboard
        unique_id: pv2_dc_input_yield
        unit_of_measurement: "kWh"
        icon: mdi:solar-power
        state: >
          {{ states('sensor.PV2_Yield_Sun') | float(0)}}
        device_class: energy
        state_class: total_increasing

sensor:
  - platform: integration
    name: "PV2 Yield Sun"
    source: sensor.pv2_power_dc
    unit_prefix: k
    unit_time: h

It seems to work. Lets have a look at a other day with sunlight.

johanvh83, 2021-12-22:
Change mine too according your idea. Had already made a template sensor based on the input_power and made also the sensor integration (sum) but couldn't insert it in the Energy integration because the last template sensor with the correct device_class and state_class was missing. I believe indeed, this should work. Keep you informed! Thanks, anyway

killee, 2021-12-23:
@johanvh83 I think you have to wait a bit.

With the integration I have another issue:
[image]
And the Power:
[image]

It seems that this is a issue with the integration...

johanvh83, 2021-12-23:
Hi, I checked actual state, but all seems to work correctly.

This is my sensor I used in the Energy integration as solar production:
[image]

So far everything looks normal and correct:
[image]

FusionSolar:
[image]

FusionSolar gives yield of  2,47 kWh till 14:05, HA gives 2,7 kWh in the Energy integration till 14:00, there is a difference. Don't know how it's possible. FusionSolar should give more than 2,7 kWh when the measurement is taken 5 minutes later. It has problably to do with some loss...

killee, 2022-01-01:
I've the same issue...

But fusion did it wrong.
The Daly yield in fusion is only the energy in the battery without the consumption that is directly used...

inigoml, 2022-01-06:
Same problem here after adding battery.
In my understanding, "Consumed Solar" includes also "Consumed Battery" and since it is added, real consumption is considered twice.

[image]

inigoml, 2022-01-07:
> I think I have solved this issue. The attribute gives the DC Power. I integrate it with HA, than template it.

I've implemented your solution and is working properly. However, there is a problem by not using total_generation counters. If for some reason you loose access to inverter (and this happens time to time, at least for me), you won't update stats for that period and there is no way to recover them.
I think a more accurate approach would be to subtract to sensor_total_yield_XXXXXX those values given by sensor.storage_total_discharge_XXXXXXX. This way, we would have ONLY energy produced by PV system and not dumped by battery.