Emilv2/huawei_solar #61: Integrating with energy dashboard

themaninspain, 2021-12-15:
I'm trying to get the energy output of the inverter to display in the energy dashboad, but only thedaily_yield sensor is available.
Really what I want to use is the "sensor.sun2000_6ktl" which gives me the instantaneous reading but the entity is not listed in the "Configure Solar Panels" and the help says to contact the developer.

JacoboSanchez, 2021-12-15:
Energy dashboard uses energy (kwh) and sensor.sun2000_6ktl is a power sensor (W), that is why it gets not listed as an option. Energy dashboard would never give you the instant W produced but the energy generated in a determined time window

themaninspain, 2021-12-15:
thanks for the clarification. So I have to use one of the yield sensors instead?

JacoboSanchez, 2021-12-15:
Yes, daily or total yield should work. My advice is to use total as daily resets to 0 at midnight but may give you issues if the system time between home assistant and inverter is not the same (if they do not reset the day at the same time). Total yield is incremental from the installation so home assistant is able to add to the current day the energy added from the last reading of the previous one

JacoboSanchez, 2021-12-15:
For reference [another post](https://github.com/Emilv2/huawei_solar/issues/56#issuecomment-986593041) for a similar question where you can see some cards to have a power flow diagram in a dashboard (not the energy one from HA)

themaninspain, 2021-12-15:
One thing I have noticed is my SUN2000 has a clamp for measuring the energy consumption of the whole home and the output I get from the HA sensors is net energy production not what the panels are producing. The Huawei solar app gives me both numbers

johanvh83, 2021-12-16:
My situation is more complex because I have a battery. I'm still struggeling with the real solar power production. It's a long story; it could be find here: https://community.home-assistant.io/t/struggling-to-setup-energy-integration-for-huawei-solar-system-with-battery/367664

themaninspain, 2021-12-21:
I've taken a look at the the other posts and managed to set up the sensors and integrate them into the energy panel.
Used two templates:

`# energy used
- sensor:
  - name: energy_used
    unit_of_measurement: 'kWh'
    state:
      "{% if states('sensor.total_yield_serial_number') | float(0) > 0 and state_attr('sensor.sun2000_6ktl_l1_serial_number','grid_exported_energy') | float(0) > 0 and state_attr('sensor.sun2000_6ktl_l1_serial_number','grid_accumulated_energy') | float(0) > 0 %}
       {{ '%0.2f' | format(states('sensor.total_yield') | float(0) - 
                           state_attr('sensor.sun2000_6ktl_l1_serial_number','grid_exported_energy') | float(0) + 
                           state_attr('sensor.sun2000_6ktl_l1_serial_number','grid_accumulated_energy') | float(0)) }}
       {% endif %}"
    device_class: energy
    state_class: total_increasing`

and

`# grid exported energy
- sensor:
  - name: grid_exported_energy
    unit_of_measurement: 'kWh'
    state:
#    value_template: >
         "{% if state_attr('sensor.sun2000_6ktl_l1_serial_number','grid_exported_energy') | float(0) > 0  %}
          {{ state_attr('sensor.sun2000_6ktl_l1_serial_number', 'grid_exported_energy') | float(0) }}
          {% endif %}"
    device_class: energy
    state_class: total_increasing`

Remember to change the sensor names to your installation (I found out the hard way :-()

themaninspain, 2021-12-22:
Yes, that's what the "device_class" and "state_class" are for.
And so far it's working well.

johanvh83, 2021-12-22:
Apologies, had to define my question better:
Your templates makes 2 new sensors: energy used and grid exported energy 
Where do you put them in the Energy setup?
Grid consumption? Return to grid? Solar production? Battery system?

themaninspain, 2021-12-22:
OK, this is how they are setup in the energy integration:

"Energy used" -> "Grid Consumption"
"grid exported energy" ->  "Return to grid"

johanvh83, 2021-12-22:
Ok, clear. I have a DSMR (digitale grid meter) so I read out the 'real' values. My problem is to get proper values for the Solar production. I have also a LUNA 2000 battery and so far I'm not able to set it up properly

themaninspain, 2021-12-22:
Take a look at the 7th post in this thread, there are links to 2 posts with more info. I used them to work out how to get the readings I needed and they discuss installations with batteries.

themaninspain, 2022-01-21:
I'm having a problem with the readings and I'm not sure if it is an issue with the readings themselves or the energy dashboard in HA.
The "Energy from grid" measurement in the Dashboard is showing a number which is the real energy consumed minus the energy returned to the grid.

themaninspain, 2022-08-02:
Hi,
with the latest version of the integration you don't need these. 
The readings from the inverter show up as "Power Meter Consumption", "Power Meter Exported" and "Inverter Daily Yield".
These can all be used directly in the Energy dashboard.

webdados, 2022-08-02:
Is it "grid_accummulated_*", "grid_exported_*" and "daily_yield_*"?
I can see them with the names you mention.

[image]

webdados, 2022-08-02:
I've used those and I wonder if I shouldn't be seeing the energy produced, sent to the grid and consumed on each interval, instead of always seeing the totals:

[image]
[image]

webdados, 2022-08-02:
I'm on 1.2.4, which is the latest.

Can you confirm those are the right entities?

webdados, 2022-08-02:
@Emilv2 Isn't there any documentation on which entities to use on the energy dashboard? I think that's a must.

themaninspain, 2022-08-02:
These are entities that give the readings used in the energy dashboard
[image]

webdados, 2022-08-02:
@themaninspain that's weird because I don't have exactly those then, as you can see from the above screenshot unless you renamed them (?)

themaninspain, 2022-08-02:
It seems that I am now using a different integration!
I had some problems after the last update and had to delete and reinstall and I'v e just realised that there are 2 Huawei Solar integrations available in HACS and I managed to install the other one.
This is the reason why the entities are different

themaninspain, 2022-08-02:
go to the HACS website and search for Huawei Solar, that is the integration I have installed

webdados, 2022-08-03:
@themaninspain What a actual mess!

themaninspain, 2022-08-03:
I hadn't realised that I'd installed a different integration until you mentioned the different entity names.
When I reinstalled only the one integration showed up,  it lives in a folder called "huawei_solar" and the integration name is the same. There's no immediate way of knowing that it is not Emil's integration. you can tell by clicking on the known issues link as it takes you to a different website.
I think that they should at least have different names so that you know which one you are installing