Emilv2/huawei_solar #63: Questions for the inveter

JensPauwels, 2021-12-19:
1) Is there a reason for the negative watt peak from the inverter?

[image]

2) There's a weird behaviour in the daily_yield. I'm currently using this energy grid setup:
[image]

However the daily_yield also shows some values at night, which is find very weird. Is it possible to explain how the daily yield works, because perhaps i'm misunderstanding it

[image]

Emilv2, 2021-12-19:
No, that's a bug. You better use total yield for the energy panel.

JensPauwels, 2021-12-19:
Thank you, the total yield indeed looks more realistic. Is there an explanation for the night yields?

Emilv2, 2021-12-19:
The negative yield around 00:00 is because the reset time is not correctly synced, I thought I coded this correctly but apparently not. I don't really understand what's going on with the night yields or the negative peak.

JensPauwels, 2021-12-19:
The host is a debian server which is synced, is it a setting in home assistant that I need to change to prevent this or will you fix this in a next version?