Emilv2/huawei_solar #64: Unable to find inverter

sasholinia, 2021-12-25:
Hi, 
Merry Christmass!
i've got a problem with the integration since the last software integration of the inverter. Before that everything was fine, but now i have this error: could not connect to Huawei inverter: failed to connect to device, is the host correct?. I can correctly use the app Solar Fusion. The last version of the inverter is the V200R001C00SPC117. How can I solve it?

Thank you!

Agimgrue, 2021-12-25:
Hi, same problem two hours ago, I think my Sun2000 5KTL has been updated and now it's imposible to connect. 
Setup of platform huawei_solar is taking longer than 60 seconds. Startup will proceed without waiting any longer.
Someone knows how we can do, if we can dow a downgrade or similar? Thanks

ph30n1x, 2021-12-29:
This integration seems broken now. Cannot connect to the invertor anymore.

Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 6:53:49 PM (9 occurrences)
Last logged: 6:54:36 PM

Failed to connect: [Errno 113] Connect call failed ('10.0.1.178', 502)

akdri, 2021-12-30:
Try downgrade the firmware
Download V200R001C00SPC114 to official Huawei page, https://support.huawei.com/enterprise/en/digital-power/sun2000l-pid-22027611/software
First register your inverter and login.
When you download de firmware, use de mobile app to upload.

maciej202011, 2021-12-30:
Hi, the same problem
*****
Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.9/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 14:26:57 (11 occurrences)
Last logged: 14:28:44

Failed to connect: [Errno 111] Connect call failed ('192.168.*.*', 502)
*****

akdri, 2021-12-30:
Firmware 117 port TCP 502 modbus is closed.
Downgrade firmware 114.
https://forum.huawei.com/enterprise/es/huawei/m/ViewThread.html?tid=806087&lang=es

ph30n1x, 2021-12-31:
Why has the port been closed? If Huawei disable features permanently I will
pursue this with consumer protection in Australia as this is unacceptable

sasholinia, 2022-01-03:
That's the answer Huawei gave me:

"Unfortunately the SUN2000L1 supports data read and write only when the FusionSolar app is directly connected to the inverter at the local end. Modbus TCP connections to devices except FusionSolar will not be supported in the future.
Dongle has committed to developing the MODBUS-TCP protocol,
In addition, if a customer needs to use a third-party device and use Modbus TCP to connect to the device, only Dongle supports this function and must comply with the Modbus TCP protocol provided by Dongle.
In conclusion with the new firmware for the model L1 if you want to use Modbus-TCP you need to install a SDongle."

So i think we have 2 options:
- downgrade the firmware
- buy a dongle

I would be happy if Emilv2 could find a different solution, but i'm not an expert so i don't know if it's possible...

JacoboSanchez, 2022-01-15:
Anyone has the apk for the fusion solar android app that works with the 113 firmware? I don't want to upgrade but now I can not connect with the app

joaovitoriasilva, 2022-01-15:
> That's the answer Huawei gave me:
>
> "Unfortunately the SUN2000L1 supports data read and write only when the FusionSolar app is directly connected to the inverter at the local end. Modbus TCP [...]

Hello. I don’t have this problem because I am on version 109, but I would like to be prepared. What dongle is that you are talking about? Thank you.

sasholinia, 2022-01-16:
> 

Hi,
as I know, there are different type of dongles: Wifi, ethernet and 4g but actually I don't have more information...

nattgris, 2022-01-16:
From the forum link above:

> Hi all, Due to security issue, we close the 502 port,
> we have below suggestion.
> 1. First, we do not advise customers to use the inverter Wi-Fi to do anything except log in to Fusionsolar app. Even it can used, we do not guarantee the stability.
> 2.Second, connecting to third parties via modbus communication with dongle is a mature solution.
> 3.Third, if the customer wants to use the local connection to their system, the port 6607 can be used. However, we are not responsible for solving the problem.

So can someone see what port 6607 does? I'm considering to buy this inverter but I need to know that I can have full local control/visibility without any disgusting cloud service.

sasholinia, 2022-01-26:
I still have not clear how to connect...should I connect the RP4 to the inverter AP and than use the 6607 port?

nattgris, 2022-01-26:
That's how I interpreted that thread. Modbus TCP should be available on port 502 via the dongle, and apparently only on port 6607 when connected directly to the inverter AP.