Emilv2/huawei_solar #71: Daily charge and recharge not resetting

joaovitoriasilva, 2022-01-19:
Hello,

Hope you are well and safe.
It appears that the "storage_total_charge" and "storage_total_discharge" entities are not resetting so I am getting strange values for these components. If I check the main entity attributes this information is correct. Attached are two images where you can see the problem.

Integration is configured like below:
- platform: huawei_solar   
    host: '192.168.X.X' #Dongle IP
    #optimizers: true
    battery: true
    slave: 1 #WiFi dongle in use

Thank you

[image]
[image]

joaovitoriasilva, 2022-01-19:
Hello again,

Daily yield is also not okay. App reports 9.25kWh.
[image]

Emilv2, 2022-01-19:
Total yields are not supposed to reset, better use those instead of the daily ones.

joaovitoriasilva, 2022-01-19:
Hi, thank you for the response.
I did not understand. So I have these entities created by the integration:
[image]

When you talk about using the daily ones, what do you mean by that? Create new entities based on sensor templates? Like below?
- name: "Solar Battery charge"
      unique_id: solar_battery_current_day_charge_capacity
      state: >
        {{ states.sensor.sun2000_4ktl_l1_.attributes.storage_current_day_charge_capacity  }}
      device_class: power
      unit_of_measurement: "kW"
      icon: mdi:battery-plus
    - name: "Solar Battery discharge"
      unique_id: solar_battery_current_day_discharge_capacity
      state: >
        {{ states.sensor.sun2000_4ktl_l1_.attributes.storage_current_day_discharge_capacity  }}
      device_class: power
      unit_of_measurement: "kW"
      icon: mdi:battery-minus
      
Finally, I can't get the daily yield from main entity attributes and that is reporting wrongly App reports 9.25kWh and entity reports 4.77.

Emilv2, 2022-01-19:
Use total yield, total charge and total discharge

joaovitoriasilva, 2022-01-19:
Hello. Thank you for the response. 
Can you be more specific? I can’t understand how.