Emilv2/huawei_solar #74: Few times total production report 0

WMP, 2022-01-26:
Hi, after a month of using your integration, the integration one day returned me a value of 0 total production: https://i.imgur.com/46fCl7t.png . At that time we had -16°C so it's possible that the network device froze, but I don't have any error logs or connection problem for that. According to the logs everything was ok. 

Unfortunately reducing the value to 0 and returning to previous value caused that the Energy panel in HomeAssistant calculated me 12000 kWh of production: https://i.imgur.com/TvxFzaM.png . 

Is it possible that the object that your integration returns to HomeAssistant has information that the value cannot be reduced? Maybe something like this https://community.home-assistant.io/t/only-allow-a-sensor-value-to-increase/298648 ?

maartenhbe, 2022-01-27:
Recently had the same situation on pretty much every sensor/attribute causing among others a false consumption spike uit to 42 million kWh in a single hour. It looked more of a temporary issue on the inverter/dongle side though I couldn't find any alarms via the FusionSolar app.

Emilv2, 2022-01-27:
If the connection is broken the values should be invalid and not 0, I'm not sure why this did not happen here. That said, I don't have much time to work on this and will not release any more updates, but @wlcrs is working on updating the library and adding the integration into HA.