Emilv2/huawei_solar #77: HA 2022.2.1 Solar yield no longer shows in the energy dashboard - (known problem)

themaninspain, 2022-02-04:
After upgrading to HA 2022.2.1 I no longer have the solar yield in the energy panel.
After doing some research I came across this post [https://community.home-assistant.io/t/energy-integration-stopped-working-in-2022-02-mysensors-fails-to-update-values-in-db/388421/2](url) which explains that the state-class for the sensor needs to be changed from "measurement" to "total increasing".
With my limited knowledge I have been unable to find a way to fix this. Tried creating a customization for the sensor but it gives an error when changing the state class.
I also thought of creating a customer sensor (I successfully did this to get the values from the inverter, but I was in "script kiddie mode" and did not have to do many changes to get it to work).
So, the question is can the integration be changed to use the new state class or can someone suggest how to create a sensor with the new class?

themaninspain, 2022-02-04:
I finally got a solution. Turned out it was easier than I thought although it took a few hours of googling.

Added this to customize.yaml
`sensor.total_yield_serial_number:
  state_class: total_increasing`

("serial_number" is the one from the inverte)
Now the only problem is that I lost most of today's readings which screws up my weekly, monthly and yearly totals

ServusNL, 2022-02-04:
Thanks for sharing your efforts! I'm experiencing the same problem. Local time is 19:49 and it's dark so I will only be able to validate the fix tomorrow, **but again, thanks for sharing**! And I hope it gets fixed in this integration.

themaninspain, 2022-02-04:
you're welcome, I have still had a couple of hours of sunlight and it seems to have worked.
Tomorrow will be the best test.
Now all I have to do is work out how I can clear out the old data, only installed my panels 6 weeks ago, so I don't mind losing the data