Emilv2/huawei_solar #79: A lot of Update of sensor.sun2000_ is taking over 10 seconds

rufik, 2022-02-14:
There are many many warnings about exceeding 10 sec limit for polling from inverter. Could something be done to get rid of this message flooding logs?
HA Core version 2022.2.6

```
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:728
First occurred: 10:44:47 (291 occurrences)
Last logged: 15:34:47
Update of sensor.sun2000_8ktl_m1_01074300_002_bt2160214315 is taking over 10 seconds
```

swiergot, 2022-02-17:
Same here. I've been using the integration for 3 days only but I'm pretty sure I didn't see those messages on the first day.

galtamur, 2022-03-01:
Same here. I noticed this Warning is shown every 6-10 minutes. The frequency is increasing when for example I save an automation, probably due to the resources capacity usage for that.

rufik, 2022-03-08:
> Logger: homeassistant.helpers.entity
> Source: helpers/entity.py:728
> First occurred: 6 march 2022, 18:16:09 (**2350 occurrences**)
> Last logged: 09:26:21
> Update of sensor.sun2000_8ktl_m1_xxxxxxxxxxxx is taking over 10 seconds

2350 occurrences in 50h, that's a lot!

drizzle1, 2022-05-28:
+1. The integration seems to pass/update values much slower than before. Or is it a HA thing? Not sure, but i'm pretty sure it wasn't always lagging behind like it has been for the past few weeks.