Emilv2/huawei_solar #80: Extreme high total_yield

tgrosz, 2022-02-18:
Hi,
Sometimes (twice a week) I see extreme high reported values in HA total_yield, like 490kWh for a day. Can you help what to check to be able to give more details?

cookalone, 2022-02-22:
I have the same problem. Looking at the history, the total yell have once on a while a communication error and produces a cero entry for a minute and comes back to the total.
I am new in home assistant but it seems to me that the error trigger a new subtotal.
[image]

rfaelens, 2022-02-25:
This issue also arises when updating to a newer version of huawei_solar.
Checking the sqlite3 database produces the following:
```
sqlite> SELECT * FROM statistics WHERE created > '2022-02-20 06:30' AND created < '2022-02-21 13:30' AND statistics.metadata_id = 46;
id          created                     metadata_id  start                       mean        min         max         last_reset                  state       sum
----------  --------------------------  -----------  --------------------------  ----------  ----------  ----------  --------------------------  ----------  ----------
162583      2022-02-20 07:00:13.741909  46           2022-02-20 06:00:00.000000                                      1970-01-01 00:00:00.000000  997.49      498.61
162663      2022-02-20 08:00:13.756545  46           2022-02-20 07:00:00.000000                                      1970-01-01 00:00:00.000000  997.53      498.65
162743      2022-02-20 09:00:13.908277  46           2022-02-20 08:00:00.000000                                      1970-01-01 00:00:00.000000  997.67      498.79
162823      2022-02-20 10:00:13.665622  46           2022-02-20 09:00:00.000000                                      1970-01-01 00:00:00.000000  997.94      499.06
162903      2022-02-20 11:00:13.827640  46           2022-02-20 10:00:00.000000                                      1970-01-01 00:00:00.000000  998.2       499.32
162983      2022-02-20 12:00:13.766409  46           2022-02-20 11:00:00.000000                                      1970-01-01 00:00:00.000000  998.47      499.59
163063      2022-02-20 13:00:14.465666  46           2022-02-20 12:00:00.000000                                      1970-01-01 00:00:00.000000  998.78      499.9
163143      2022-02-20 14:00:13.925521  46           2022-02-20 13:00:00.000000                                      1970-01-01 00:00:00.000000  999.04      500.16
163223      2022-02-20 15:00:13.787399  46           2022-02-20 14:00:00.000000                                      1970-01-01 00:00:00.000000  999.29      500.41
163303      2022-02-20 16:00:13.881809  46           2022-02-20 15:00:00.000000                                      1970-01-01 00:00:00.000000  999.36      500.48
163383      2022-02-20 17:00:13.844601  46           2022-02-20 16:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163463      2022-02-20 18:00:13.909834  46           2022-02-20 17:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163543      2022-02-20 19:00:13.862804  46           2022-02-20 18:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163623      2022-02-20 20:00:14.001150  46           2022-02-20 19:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163703      2022-02-20 21:00:14.141870  46           2022-02-20 20:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163783      2022-02-20 22:00:14.154288  46           2022-02-20 21:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163863      2022-02-20 23:00:14.119896  46           2022-02-20 22:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
163943      2022-02-21 00:00:15.554695  46           2022-02-20 23:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164023      2022-02-21 01:00:14.120698  46           2022-02-21 00:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164103      2022-02-21 02:00:14.126154  46           2022-02-21 01:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164183      2022-02-21 03:00:14.209046  46           2022-02-21 02:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164263      2022-02-21 04:00:14.158456  46           2022-02-21 03:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164343      2022-02-21 05:00:15.888398  46           2022-02-21 04:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164423      2022-02-21 06:00:14.162388  46           2022-02-21 05:00:00.000000                                      1970-01-01 00:00:00.000000  999.37      500.49
164503      2022-02-21 07:00:14.216980  46           2022-02-21 06:00:00.000000                                      1970-01-01 00:00:00.000000  999.38      500.5
164583      2022-02-21 08:00:14.269441  46           2022-02-21 07:00:00.000000                                      1970-01-01 00:00:00.000000  999.99      501.11
164663      2022-02-21 09:00:14.197177  46           2022-02-21 08:00:00.000000                                      1970-01-01 00:00:00.000000  1000.67     501.79
164743      2022-02-21 10:00:10.580166  46           2022-02-21 09:00:00.000000  1000.74     1000.74     1000.74                                 1000.87     0.11000000
164840      2022-02-21 11:00:10.638103  46           2022-02-21 10:00:00.000000                                                                  1001.41     0.64999999
164941      2022-02-21 12:00:10.610276  46           2022-02-21 11:00:00.000000                                                                  1001.64     0.87999999
```

As you can see, the last_reset was adapted with the new version of huawei_solar. Find the metadata_id for your huawei-solar entities by doing `SELECT id, * FROM statistics_meta WHERE statistic_id LIKE '%10214004%';` and then:
```
UPDATE statistics SET last_reset=NULL,sum=state WHERE metadata_id=49;
```

This solved the problem on my installation.

Rvh91, 2022-03-02:
I'm having a similar issue, but for me it does not report a zero value, but a small non-zero value instead. For different sensors I was using a template to remove these values to be used in the energy dashboard. However, since the low reported value is not zero nor is it repeatable (see bellow). I cannot use the same approach here. Can anyone comment on how to solve this? 

I have no experience with checking the database or with editing the database, so I'm not confident I could apply the suggestion from the comment above.

[image]
[image]

rufik, 2022-03-02:
Same here, sensor gets 0 for a single reading and gets back to proper value. Maybe there should be some "protection check" in the code that converts this zero to "unknown" state?

JensPauwels, 2022-03-02:
All the values are indeed super high after updating to the latest huawei solar.

See the image below

[image]

rufik, 2022-03-08:
> Logger: homeassistant.helpers.entity
> Source: helpers/entity.py:728
> First occurred: 6 march 2022, 18:16:09 (**2350 occurrences**)
> Last logged: 09:26:21
> Update of sensor.sun2000_8ktl_m1_xxxxxxxxxxxx is taking over 10 seconds

2350 occurrences in 50h, that's a lot!

rufik, 2022-03-14:
I can confirm value is zero.

meceo, 2022-03-19:
I try to fix the issue with a template with following code:

```
  - name: "Instalacja 11,6kW Całkowita Produkcja"
    unique_id: plant_11_6_total_production
    unit_of_measurement: "kWh"
    state: >
      {% if states('sensor.total_yield_hvXXXXXX') | float(0) > 0 %}
      {{ '%0.2f' | format(states('sensor.total_yield_hvXXXXXX') | float(0)) }}
      {% endif %}
    device_class: energy
    state_class: total
```

So far so good for the last several days.

gilder0, 2022-04-10:
Recently, I have the same problem once a week or less.
But earlier, when I've used HA 2021.12 i never had problems

In my opinion problem may be due to the fact that this integration creates independent entities working asynchronously and retrieve data from one common entity for the inverter.
I think it sould be rewriten to use the DataUpdateCoordinator mechanism.

swiergot, 2022-04-22:
Same here. Until recently it happened very rarely, now it happens a few times a day.

meceo, 2022-04-22:
> I try to fix the issue with a template with following code:
>
> ```
>   - name: "Instalacja 11,6kW Całkowita Produkcja"
>     unique_id: plant_11_6_total_production
>     unit_of_measurement: "kWh" [...]

I can confirm that the fix above works for me. Since 19th March I don't have the issue.

rufik, 2022-04-22:
This is not a fix, it's just workaround.

WMP, 2022-05-21:
This can be fix, but I don't test this: https://github.com/wlcrs/huawei_solar/issues/39