Emilv2/huawei_solar #82: Not capable of connecting

Rvh91, 2022-02-28:
Hi, 

First of all thank you for making this integration. Would love to have the inverter integrated in HA.
Not sure if this is the right place to reach out, but I cannot seem to setup this integration. 

My setup is as following:
Huawei SUN2000-8KTL-M1, firmware: SPC140
     Modbus-TCP → Enable(unrestricted)
     Parallel system communication parameter setting → Parallel communication mode" to RS485
Huawei SdongleA-05, firmware: SPC124, 
     connected via UTP cable to my home network with IP: 192.168.178.52/24.

I've added the following in my configuration:
```
sensor:
  - platform: huawei_solar   
    host: '192.168.178.52'
    optimizers: false
    battery: false
    slave: 1
    port: 6607
```

I'm using the IP of the Sdongle for the host. But no sensors are showing up. I also don't see anything in the logs in relation to this integration. I'm not sure how to proceed in debugging. Could someone provide me some direction?

Rvh91, 2022-02-28:
I'm now getting some entries in the log:

2022-02-28 11:42:31 ERROR (MainThread) [huawei_solar.huawei_solar] failed to connect to device, is the host correct?
NoneType: None
2022-02-28 11:42:31 ERROR (MainThread) [custom_components.huawei_solar.sensor] could not connect to Huawei inverter: failed to connect to device, is the host correct?
...
2022-02-28 11:42:31 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:42:32 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:42:38 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
...
2022-02-28 11:42:33 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:42:44 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:42:57 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:43:23 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
...
2022-02-28 11:44:14 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)

Rvh91, 2022-02-28:
Setup of sensor platform huawei_solar is taking over 10 seconds.
11:42:38 AM – (WARNING) Sensor
could not connect to Huawei inverter: failed to connect to device, is the host correct?
11:42:31 AM – (ERROR) huawei_solar (custom integration)
failed to connect to device, is the host correct?
11:42:31 AM – (ERROR) /usr/local/lib/python3.9/site-packages/huawei_solar/huawei_solar.py

2022-02-28 11:42:28 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:30 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:31 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:32 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:33 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:35 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:38 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:44 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:42:57 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:43:23 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:44:14 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:45:56 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)
2022-02-28 11:49:21 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.178.52', 6607)

nunocampos105, 2022-03-02:
Hello,

Hope everyone is ok!

First, thanks for your work !

I've got a similar problem to thisuser, i've install the integration via HACS, but once i try to connect the system to my home assistant it says "fail to connect".

I connect the inverter via WIFI, and is on the same network as the NUC.

Huawei SUN2000-3KTL-M0,

    SW: V200R001C00SPC117

I didnt change the port (502) and i put the internal IP address.

Can someone give me some light ? :)

Thanks in advanced,

Best Regards,
Nuno

Rvh91, 2022-03-03:
So if you access the internal wifi directly. I think you have a minimum requirement for the firmware version, but also a maximum. I recall that i read somewhere that Huawei dropped the modbus support when connecting to the internal wifi directly. You could try to use a different port, since the required port can be different for different firmware versions. I have seen at least 6607 listed somewhere as a potential port. You could maybe contact Huawei to ask them if your firmware version supports modbus on the internal wifi and what port to use?

Otherwise I Expect you need a smart dongle (just google for Huawei SdongleA-05), but these are pretty expensive.