Emilv2/huawei_solar #85: Integration stops working

Steven1994, 2022-03-25:
I have the integration running for a few months now and now is the third time that it stops working, last time I waited for a couple of days and then it started working again without me doing something to fix it. 

Is there any know problem that looks like this, or is it an option to look into it, if yes, please tell me which information i should provide to do the troubleshooting. 

Best regards, 

steven

scoggins, 2022-03-26:
Check the version of firmware on your inverter. If your inverter has been upgraded recently this will be the problem. It appears Huawei have disabled the MODBUS port on the home wifi connection in firmware newer than 114.  See this thread:

https://forum.huawei.com/enterprise/en/cant-connect-to-inverter-modbus-tcp-v200r001c00spc117-sun2000-6ktl-l1/thread/816679-100027

webdados, 2022-07-02:
@scoggins Any solution for this?

scoggins, 2022-07-02:
> @scoggins Any solution for this?

I set up a raspberryPi that connects to the wifi SSID SUN2000-HSV... and has nginx stream proxy for the modbus port onto my home network via Ethernet.

Only solution I've found so far.

webdados, 2022-07-02:
Oh my... that's just a big workaround, which probably also means that each time you want to connect to the inverter via the Soler Fusion app to do some settings, you need to disconnect that raspberryPi, right? Also, it works until they decide there's no Modbus at all available via their own wifi.
Is it still sending information to their cloud?

webdados, 2022-07-02:
Link for my own reference: https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#getting-connectivity-between-ha-on-your-home-network-and-the-inverter-ap

scoggins, 2022-07-02:
Yeah. If you want to go via device commissioning. You need to disable the nginx proxy or it won't let you connect.

webdados, 2022-07-02:
@scoggins Any tutorial to do it? I have a Rpi1 and a Rpi3 layin' around.

scoggins, 2022-07-02:
Don't really have a tutorial. I'd imagine there is one out here somewhere. But the short run down of what I did was:

1. Install ubuntu 20.04 server onto a raspberry pi 3 sd card 
2. Configure /etc/netplan/50-cloud-init.yaml to have something like this to setup the bridge between your networks:
```yaml
network:
    ethernets:
        eth0:
            dhcp4: true
            optional: true
    version: 2
    wifis:
        wlan0:
            access-points:
                "SUN2000-HV<serial>":
                    password: "<password>"
            dhcp4: yes
            dhcp6: yes
            optional: true
```
3. Run **apt-get install nginx**  to install nginx
4. Create /etc/nginx/modules-available/huawei-inverter.conf which contains:
```nginx
stream {
	server {
		listen			0.0.0.0:6607;
		proxy_pass		192.168.200.1:6607;
		proxy_connect_timeout 	2s;
		proxy_timeout		30s;
	}
}
```
5. Enable that module running **ln -s /etc/nginx/modules-available/huawei-inverter.conf /etc/nginx/modules-enabled**
6. Run systemctl enable --now nginx

That should give you a nginx thats running a stream proxy for port 6607. You should then be able to point home assistant at the IP of your raspberry pi on your home network (I normally reserve the IP in DHCP or you can change (2) so that the netplan sets a static IP [Netplan Config Examples](https://netplan.io/examples/#using-dhcp-and-static-addressing).

scoggins, 2022-07-25:
Yes, you need to follow the instructions at https://forum.huawei.com/enterprise/en/modbus-tcp-guide/thread/789585-100027 that show how to enable the modbus port again. Firmware above 115 disable it by default and you need to re-enable it.

scoggins, 2022-08-01:
I've been having some issues with this setup and the proxying of the connection would just stop. The end solution I came up with was I now have a raspberry PI running ubuntu that has ethernet to my home network, and connects to the WIFI SUN2000 of the inverter. I've the configured the rPI Ubuntu box to be a gateway and to NAT (via iptables) the traffic to the SUN2000xxxx wifi network. Then I added a static route on my router to send 192.168.200.0/24 to the rPI. This means that my HA box can now talk directly to 192.168.200.1 without the need for a proxy. It also means the inverter can't talk back into my home network via it's internal interface.