Emilv2/huawei_solar #87: Huawei connection to LAN only + suitable HW

MEKadan, 2022-04-21:
I've been searching for information about the way to connect Huawei inverters to LAN and HA via LAN

I've not ordered the solar power system yet, but wish to know few essential elements about connecting the device

- Is there a in-build way to connect the device to net? The brochures I could find from system vendors don't seem to mention that?
- There is a dongle SDongleA-05, that seems to connect to wifi. Is this required / any other hw that can be used? Local availability of the dongle seems to be very poor.
- Most importantly: Can Huawei inverters be used in LAN only without connecting to any Chinese servers whatsoever?

MEKadan, 2022-05-31:
Anything?

madimadi, 2022-06-01:
Hi,
Based on my experience over the past few months:
I have SUN2000-10KTL-M1 it comes with a SDongle which has wifi and ethernet interface.
Local connectivity is possible, I managed to collect every metrics and status info via modbus tcp from it using local connection.
I believe it works "offline" without issue, I'm planning to block its MAC address on my router soon.

krzakx, 2022-07-04:
> Hi, Based on my experience over the past few months: I have SUN2000-10KTL-M1 it comes with a SDongle which has wifi and ethernet interface. Local connectivity is possible, I managed to collect every [...]

Could you check your firmware versions?

MEKadan, 2022-07-04:
Hi @madimadi I would be very interested in how you do this. Can you pls. provide some more ddetailed instructions

madimadi, 2022-07-04:
You need to set modbus enable by directly connecting to inverter wifi signal via app. Afterwards the modbus over tcp begin to answer on tcp/502. I use wifi connection at the moment, so dongle has no wired ethernet.  Still I believe it should work on LAN intrrface the same way, too. Check forum threads or more details.

MEKadan, 2022-07-04:
@madimadi I have modbus enabled, and I have tried ports 502 and 6607 to no avail.

I feel very essential, that I do not mix my home network with chinese brand hw, so I have configured a wireles bridge (router in client mode) which tries to contact inverterter wifi ap at 192.168.200.1 and my home network router connects to Huawei via this wireless bridge with static route. 

The method is described here: https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter#getting-connectivity-between-ha-on-your-home-network-and-the-inverter-ap=

Problem is that it does not respond to anything. Pings, traceroutes etc.