Emilv2/huawei_solar #89: Can't connect to inverter (port 502)

stevenvw1973, 2022-06-10:
I'm having trouble getting this to work. I use this as my config:

sensor:
  - platform: huawei_solar   
    host: '192.168.123.133'
    battery: true

And I see this in my loggings:

2022-06-10 14:57:31 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:32 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:34 ERROR (MainThread) [huawei_solar.huawei_solar] failed to connect to device, is the host correct?
NoneType: None
2022-06-10 14:57:34 ERROR (MainThread) [custom_components.huawei_solar.sensor] could not connect to Huawei inverter: failed to connect to device, is the host correct?
2022-06-10 14:57:34 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:36 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:37 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:39 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:40 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform huawei_solar is taking over 10 seconds.
2022-06-10 14:57:42 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:57:48 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:58:01 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:58:04 WARNING (MainThread) [homeassistant.components.netatmo] Webhook not registered - https and port 443 is required to register the webhook
2022-06-10 14:58:27 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)
2022-06-10 14:59:18 WARNING (MainThread) [pymodbus.client.asynchronous.async_io] Failed to connect: [Errno 111] Connect call failed ('192.168.123.133', 502)

I know they changed the installer password of the inverter upon installation. Could that be an issue?

krzakx, 2022-07-04:
It more probably because of blocked port for Modbus on newest frimware. See info here https://github.com/wlcrs/huawei_solar/wiki/Connecting-to-the-inverter but I do not understand yet if this is another integration or only separate repo.

WongGendheng, 2022-07-11:
You need to enable modbus tcp in the inverter with admin (installer) permissions

webdados, 2022-08-02:
And update firmware on both the Inverter and Dongle. The latest were working: https://github.com/Emilv2/huawei_solar/issues/91#issuecomment-1202739599

shagadeli, 2022-08-12:
> And update firmware on both the Inverter and Dongle. The latest were working: [#91 (comment)](https://github.com/Emilv2/huawei_solar/issues/91#issuecomment-1202739599)

Any chance you could share the firmware files? I can't really find a way to download them from Huawei.

webdados, 2022-08-12:
Sure: https://we.tl/t-qvMdmK3jtP

krzakx, 2022-10-13:
Hello again.

No clue why.... it was working and doesn't not work any more.
My Dongle Frimware is V100R001C00SPC130
and inverter Frimware is V100R001C00SPC146

```
Logger: pymodbus.client.asynchronous.async_io
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/asynchronous/async_io/__init__.py:307
First occurred: 15:05:54 (6 occurrences)
Last logged: 15:06:13

Failed to connect: [Errno 113] Connect call failed ('192.168.xx.xx', 6607)
```
Also same fail for port 502 number. 
Could you please advice me how I could try to "ping" Inverter Dongle from HA console? As I have no access to router, but in general in FusionSolar app I do see current working parameters so Inverter itself is connected to my WiFi network.